pycom/pycom-micropython-sigfox #170: fipy flash error

AKROF, 2018-06-07:
Please include the following information when submitting a bug report:

* FiPy
* Trying to update firmware via USB/UART.

When trying to flash my device. It just gives time out error.
Also tried with Pycom upgrade tool (both on ubuntu and Mac OS). Still got the same error.
The error looks like:
```
Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Entering flash mode
Flashing project
python /home/user/esp/pycom-esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before no_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 build/FIPY/release/bootloader/bootloader.bin 0x8000 build/FIPY/release/lib/partitions.bin 0x10000 build/FIPY/release/fipy.bin
esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
application.mk:627: recipe for target 'flash' failed
make: *** [flash] Error 2
```

Thank you!

Xykon, 2018-06-07:
/dev/ttyUSB0 is usually the Expansion Board 2.1 did you install the jumper wire to put the module into firmware update mode? https://docs.pycom.io/chapter/gettingstarted/installation/firmwaretool.html

AKROF, 2018-06-07:
Yes I did. We also checked over serial communication and output looks like
'''
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
'''

robert-hh, 2018-06-07:
Is the updater the only program which is using /dev/ttyUSB0?

AKROF, 2018-06-07:
yes. I did get the error 'serial port is busy' , when I have other program using same port.
The ESP connection error is not just from upgrader. I got the same error when I tried on Terminal with ESP-IDF.

ESP-IDF_PATH being ```/home/user/esp/pycom-esp-idf```

Xykon, 2018-06-07:
Do you have correct user permissions for /dev/ttyUSB0? You might need to run `usermod -a -G dial out $USER` and then logout or reboot your Linux PC.

AKROF, 2018-06-07:
I did try it with no desired output.

AKROF, 2018-06-07:
I could not type anything at REPL . but the TX and RX pins are connected correctly , I checked it twice.

Xykon, 2018-06-07:
I have checked one of our production FiPy and it works fine with an FTDI module and the pre-flashed firmware. I suggest you check the USB cable and make sure that the USB port / FTDI module provide enough current for the FiPy.

AKROF, 2018-06-07:
How much Current (Ampere) does it require.  I have given 5V connection from FTDI to FiPy.

robert-hh, 2018-06-07:
In average 250 mA, with peaks at ~500mA. So a normal USB is OK. The 5V of an FTDI should be the direct supply from USB. Programming does not require more power than normal us. Maybe less, because WiFi, Lora and LTE are not active at that time.

robert-hh, 2018-06-07:
Which terminal emulator are you using. Flow control should be switched off. Minicom sometimes makes problems. Picocom works nice.

AKROF, 2018-06-08:
I tried both minicom and picocom, but I am not able to type at  REPL on both of them. 
So I did not use any emulator , directly set the baudrate and port on terminal.

robert-hh, 2018-06-08:
With terminal emulator I mean apps like Picicom. So it look like no RX is seen by the FiPy. Do you have means to see, if theer is activity at P0, when you type in picocom, to tell, whther it's SW or HW. Even a multimeter should show some fluctuation away from ~3.3 V when you type or just paste a long string.

AKROF, 2018-06-08:
Yes I understood what you meant by emulator.
I have tried using picocom and I got the following result when connected with FiPy. I couldenot give or type anything after that. 
```usr@usr-ThinkPad-T460s:~$ picocom -b 115200 -r -l /dev/ttyUSB0
picocom v2.2

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : yes
nolock is      : yes
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Type [C-a] [C-h] to see available commands

Terminal ready
MicroPython v1.8.6-834-g5677703c on 2017-11-13; FiPy with ESP32
Type "help()" for more information.
>>> 
```
The main.py is with LoRaWAN program updated.

robert-hh, 2018-06-08:
If main.py contains code, that is executed and does not terminate, then you will not get a REPL response. The firmware will enter REPL as soon as main.py finishes. But that does not affect the problems with firmware updates. For that, you should check if there are level transitions on P0 if you type something in picocom.

Xykon, 2018-06-08:
You can try using [safeboot](https://docs.pycom.io/chapter/gettingstarted/programming/safeboot.html) to avoid executing boot.py and main.py

If the module comes up with the default wifi access point, you can also try to connect via wlan with telnet or ftp to remove the boot.py and main.py

If that works we can do more diagnostics from there.

AKROF, 2018-06-11:
I was able to boot it with FTP. Is ist possible to update the firmware over the air instead of serial communication

AKROF, 2018-06-11:
Hello guys, 
Thank you so much for the suggestions. I was able to update my device. All I had to do was change my USB/UART converter.

robert-hh, 2018-06-11:
Very good, and have fun with your device. You may then close this issue. By the way, questions like yours are better asked in the forum.

AKROF, 2018-06-11:
Oh ya sure. Will make sure to post it in Forum next time. Thanks again

closed 2018-06-11 by AKROF