pycom/pycom-micropython-sigfox #171: error using make BOARD=LOPY4 TARGET=boot

HelalBR, 2018-06-09:
Hello guys,

I'm having some troubles trying to compile my own firmware. I followed every step in this tutorial: 
https://github.com/pycom/pycom-micropython-sigfox
but when i run MAKE BOARD=LOPY4 TARGET=boot i get the following error:

Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC bootloader/flash_qio_mode.c
In file included from /root/pycom-esp-idf//components/esp32/include/rom/ets_sys.h:21:0,
                 from /root/pycom-esp-idf//components/log/include/esp_log.h:21,
                 from bootloader/flash_qio_mode.c:17:
bootloader/flash_qio_mode.c: In function 'enable_qio_mode':
bootloader/flash_qio_mode.c:179:66: error: 'EFUSE_RD_CHIP_VER_RESERVE_S' undeclared (first use in this function)
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/root/pycom-esp-idf//components/soc/esp32/include/soc/soc.h:189:32: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                ^
bootloader/flash_qio_mode.c:179:66: note: each undeclared identifier is reported only once for each function it appears in
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/root/pycom-esp-idf//components/soc/esp32/include/soc/soc.h:189:32: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                ^
bootloader/flash_qio_mode.c:179:66: error: 'EFUSE_RD_CHIP_VER_RESERVE_V' undeclared (first use in this function)
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/root/pycom-esp-idf//components/soc/esp32/include/soc/soc.h:189:44: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                            ^
../py/mkrules.mk:47: recipe for target 'build/WIPY/release/bootloader/flash_qio_mode.o' failed
make: *** [build/WIPY/release/bootloader/flash_qio_mode.o] Error 1

I tried with different boards (WIPY, LOPY etc) and I get the same error. Any ideas of what is causing this?

Thanks in advance.

robert-hh, 2018-06-09:
Did you check out the proper version of esp-idf? In the esp-idf directory, issue the command

git checkout 4eab4e1b0e47c73b858c6b29d357f3d30a69c074

Xykon, 2018-06-09:
Please see [this forum post](https://forum.pycom.io/topic/3326/new-firmware-release-v1-18-0)

HelalBR, 2018-06-09:
@robert-hh , I didn't check. i issued the command and everything works fine. Thanks for the fast reply.

root@alanhelal-virtual-machine:~/pycom-esp-idf# git checkout 4eab4e1b0e47c73b858c6b29d357f3d30a69c074
M	components/aws_iot/aws-iot-device-sdk-embedded-C
M	components/bt/lib
M	components/esp32/lib
M	components/esptool_py/esptool
HEAD is now at 4eab4e1b esp32: Place gettimeofday inside IRAM.
root@alanhelal-virtual-machine:~/pycom-esp-idf# 

@Xykon I didn't notice this command. I was used to do everything just following the directions on this repository. Thanks for the clarification.

Problem solved.

robert-hh, 2018-06-12:
If the problem is solved, may I ask you to close the issue?

Xykon, 2018-06-12:
For anyone else reading this... we have reverted the master branch of pycom-esp-idf back to the correct commit needed to build the stable firmware.

closed 2018-06-12 by Xykon