pycom/pycom-micropython-sigfox #173: Decrease wifi scan time

emilxp, 2018-06-20:
Hello,

I have limited experience with system programming. I need to decrease the wifi scan time by probably modifying the firmware drivers or some middleware between the drivers and the WLAN micropython interface. I looked around the code, but I'm lost. I don't think I can do anything like that from the driver. I was unable to find the network manager code or whatever middleware there is. Could someone point me to the files that may be of interest to me for solving this problem?

Regards,
Emil Georgiev

iwahdan88, 2018-06-20:
Hello @emilxp ,

The `WLAN.scan` method of WLAN class, uses the default scan configuration where an Active scan is performed on all channels (14 channels in case of configured country is china which is the default ) to collect available APs , the default max scan time currently used is 120 ms per channel.

we are currently working on improving the WLAN class to include more options for configuring more wifi parameters.

In the meantime if you would like to decrease the scan time you can update function `wlan_scan` in `esp32/mods/modwlan.c` to pass a configuration structure to `esp_wifi_scan_start` with lower `scan_time.active.max` and putting ssid/bssid = 0
See also that [Link](http://esp-idf.readthedocs.io/en/latest/api-guides/wifi.html)

Another way is to use fast scan option where you specify in the config structure passed to `esp_wifi_scan_start` the ssid/bssid of a specific AP you are interested in, then the scan will be finished once this AP is found potentially decreasing scan time.

Thanks
I Wahdan

emilxp, 2018-06-21:
Thank you @iwahdan88 ! I'm going to look into it the following days. For now I'm going to close this issue.

closed 2018-06-21 by emilxp