pycom/pycom-micropython-sigfox #174: Development branch 1.19.0.b2: ftp shows duplicate file name

robert-hh, 2018-06-20:
Firmware 1.19.0.b2, device WiPy3
When requesting a file list with ftp, some files names are listed twice. That does not happen with uos.listdir() on the device. Ftp clients tested are command line ftp in Linux, FileZilla and FireFTP
When creating, renaming and deleting  files on the device, the name that is shown twice changes.
Maybe deleted files are not properly handled (=skipped) by the ftp server.

husigeza, 2018-06-21:
Hello,

Could you share a detailed example of steps you did? Unfortunately I could not reproduce it on my own,  no files are listed twice with FileZilla (v3.28.0)/ftp Linux command, even the delete/rename/add operation issued on the device itself or through FTP. 

Do you notice this behavior on an attached SD card too? 

Please also share the FTP settings you used for connecting.

Thanks!

robert-hh, 2018-06-21:
The ftp settings are standard, the same as I use for all Pycom  board: passive, 1 session per link. Note, that this also happens with command line ftp.
It does not happen with v1.18.0 and, and I do not use an SD card. It's on internal flash. What I did not do yet is wiping out the device, re-flash and create a new file system.
The store/delete/rename cycle happens with my onboard editor (https://github.com/robert-hh/Micropython-Editor), which, when saving
- stores the saved file to a temporary name ("<old_filename>.pytemp")
- deletes the old file name
- renames the new file to the initial file name
 That mechanism is quite generic. Whenever I do that, a different file name is shown twice. 
But I can start over from a fresh device this night. Mayby teh files system is just corrupted for a different reason.

robert-hh, 2018-06-21:
OK. I re-tested that, I took the WiPy3, 
- erased it with 'make erase', 
- installed a fresh copy of the development version with the erase flag set
- and then started to upload files with FireFTP. 
After a upload a few file, like after the 5th file, two of the existing file names were shown twice. Listing files at the REPL promt with uos.listdir() does not show this effect.

husigeza, 2018-06-22:
Hello,

Based on your detailed steps I could reproduce the issue, thanks! 
For me the problem only exists when the files are copied onto /flash directly. If they are placed in a folder (e.g.: /flash/folder) they are not listed twice, so until I fix this issue I would recommend putting your files into a folder :)

husigeza, 2018-06-22:
I did further testing... **The problem appears, no matter of the location**, when more than 8 objects (dirs+files) are located in a folder. If you copy/create the 9th file/folder then 2 items will be listed twice.

husigeza, 2018-06-22:
Hello,

The root cause was found and corrected, will be part of the next Development release. Thanks for reporting the problem!

robert-hh, 2018-06-22:
Thanks. It seems anyhow to be a  display only issue.

closed 2018-06-22 by robert-hh