pycom/pycom-micropython-sigfox #175: Socket module does not expose standard micropython constants

MoeweX, 2018-06-24:
The socket module included with the Wipy 3.0 firmware does not expose port specific constants the way other micropython ports are exposing them.

This is an issue when trying to use a micropython module such as [slimDNS](https://github.com/nickovs/slimDNS), which therefore does not run on the Wipy 3.0. More details can be found in the [issue](https://github.com/nickovs/slimDNS/issues/2) I had opened in the slimDNS repository.

Can you update your included socket implementation so that it is compatible with modules from the community that depend on it?

Xykon, 2018-06-24:
Can you please clarify the hardware? In the other ticket you linked here the discussion is about the cc3200 based WiPy while here you're talking about the esp32 based WiPy3.

Xykon, 2018-06-24:
If you're indeed looking for a solution for the WiPy3 might I suggest this DNS server: https://github.com/jczic/MicroDNSSrv (and there is also a web server available).

This DNS server is written specifically for our esp32 based (WiPy 2/3, LoPy 1/4, SiPy, GPy, FiPy) modules.

MoeweX, 2018-06-24:
Hi,

I think nickovs looked into the wrong repository, I am using the WiPy 3 (so esp32) and my issue is also related to that hardware.

I am interested in the multicast DNS aspects of the slimDNS project, i.e. getting the IP address of other devices based on their hostname, and enabling other devices to do the same for the Wipy board. This functionality is not offered by your suggestion, or is it?

Nevertheless, I think one really cool aspect about the WiPy is that it is build upon micropython, which increases its community significantly. So I think exposing the same interfaces to hardware functionality of the WiPy esp32 port is a good think.

MoeweX, 2018-07-03:
Hi @iwahdan88, I see you got assigned to this issue. Are you considering to add such a compatibility or what are your thoughts on this?

iwahdan88, 2018-07-03:
Hi @MoeweX , sorry for the late reply.

I've looked in to that issue , and I see your point, I've already added this to our backlog.
 we are currently doing updates and adding new features in bt module for better configurability and performance. 
I will leave this issue open and will close it once the updates are done.

Thanks
Cheers

husigeza, 2018-08-07:
Hello,
Thanks for reporting, the mentioned constants are added and will be part of the next development version thus closing this issue.

closed 2018-08-07 by husigeza