pycom/pycom-micropython-sigfox #176: Bluetooth

innoagg, 2018-06-28:
Pycom MicroPython 1.18.0 [v1.8.6-849-046b350] on 2018-06-01; GPy with ESP32

The Bluetooth implementation is missing some critical features that make peripheral implementations difficult. The items listed below are not all of them, but cover the most important at least from my perspective in order of importance.

Advertising
1. No way to set advertisement flags with **set_advertisement**. For example, the flags ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT are hard coded.
2. Advertisement intervals are hard coded. 
3. No way to configure a scan response.

Peripheral Connections
1. The callback for CLIENT_CONNECTED / CLIENT_DISCONNECTED events is missing any information about the client such as Mac address or some kind of connection handle. Connection management is impossible without. All callback should identify which client it is for. 
2. Method to disconnect a particular connected client. There appears to be a method **disconnect_client** but it is not clear which connection is terminates if there are multiple active.
3. Bonding or encrypting the BLE connection does not appear to be supported.

iwahdan88, 2018-07-02:
Hello @innoagg ,

Currently we are working on updating Wifi and BLE modules for more configurability and new features.
Thanks for pointing out these points, I have them in the Backlog.
I expect these updates to be added by mid of this month.

Thanks
Cheers

innoagg, 2018-07-23:
Any news on timing for these BLE items. I did not see any changes in the latest release.

iwahdan88, 2018-07-23:
Hi @innoagg , BLE updates are still not released , we are working on releasing these updates as soon as possible, sorry for the Delays.

innoagg, 2018-07-24:
Thanks, Eagerly awaiting as I have found couple more items I hope are addressed...

Characteristics defined on the pycom device as PROP_WRITE only can in fact be read successfully by a client. The pycom device should send back an indication that it cannot be read (normally 0x0403)

Also, perhaps I am not seeing how, but one should be able to send back a response code to a client connection in response to a characteristic write operation to indicate success or to validate values.

innoagg, 2018-10-23:
Found another BLE implementation shortcoming. Hardcoded scan interval and window is a big issue for us. Can't achieve 100% scan duty cycle. 

What BLE enhancements will be included with the 1.19 (soon?) release?

gretel, 2019-05-08:
it's getting disappointing in terms of what is sold and actually works.