pycom/pycom-micropython-sigfox #177: Random restarts with SiPy in 1.18.0

limartin1, 2018-07-03:
Hello,

I have a stability problem with my SiPy that I can not explain. It restarts without explanation and randomly.

First here is the firmware information:
(sysname='SiPy', nodename='SiPy', release='1.18.0', version='046b350-dirty on 2018-07-03', machine='SiPy with ESP32', sigfox='1.0.1')

I compile the firmware with my .py files in the frozen zone. I am using the latest stable version of pycom-esp-idf (1.18.0). So there is really no reason for my problem to come from here.

To try to understand my problem I log absolutely everything that happens and the consumption of RAM. I have no errors in my logs and the memory is at least 21680 dull. No problem with memory management I think.

I also log "machines.RESET_CAUSE" and there is a machine.WDT_RESET. Except that I do not use a watchdog in my program. So I do not really know why I have this cause.

For information, the program contains 3 threads that run in a loop and several functions launched regularly by timers.

Is it possible to know the CPU load to see if it can come from that? Or do you have an idea that will explain the random restart?

Thank you very much in advance,

Lilian

Xykon, 2018-07-03:
Hello Lilian,

Can you please either post your code (or at least the part of the code causing the issue) or send it via email to support (at) pycom.io with a reference to this issue?

Thanks a lot.

Regards, Christian

limartin1, 2018-07-11:
OK I think it's my fault, I had some mutex issues, so I have modify my program and all is OK ! 

Thanks you, 
Lilian

closed 2018-07-11 by limartin1