pycom/pycom-micropython-sigfox #178: LoRa constructor does not set power_mode

campbellsinclair, 2018-07-09:
* System configuration:
  * Pycom board: LoPy1.0r (with deep sleep shield DS v1.0)
  * Firmware version: (sysname='LoPy', nodename='LoPy', release='1.17.5.b6', version='v1.8.6-849-56d00234 on 2018-05-18', machine='LoPy with ESP32', lorawan='1.0.2')
  * 5V power supply
  * LoRa antenna connected.
  * Interface with the LoPy using a USB to UART adaptor.
  * pycom.heartbeat_on_boot(False)
  * main.py is:
```python
print('Hello')
```
* Exact steps to cause this issue:
  * Reset the LoPy using the hardware push-button.
  * Connect to the LoPy using UART and Visual Studio Code
  * Invoke the following commands using the micropython interpreter:
```python
from network import LoRa

lora = LoRa(mode=LoRa.LORA,
            region=LoRa.AU915,
            frequency=925500000,
            tx_power=5,
            bandwidth=LoRa.BW_500KHZ,
            sf=7,
            preamble=8,
            coding_rate=LoRa.CODING_4_5,
            tx_iq=False,
            rx_iq=False,
            public=False,
            power_mode=LoRa.SLEEP,
            )
```
*
  * With a multimeter connected in series with the 5V power supply, observe that the current is approximately 103 mA.
  * Invoke the following command:
```python
lora.power_mode(LoRa.SLEEP)
```
*
  * Observe that the current is approximately 93 mA.

campbellsinclair, 2018-07-09:
My guess is that power_mode is set in, but then another command is called which requires power_mode to be changed.

campbellsinclair, 2018-07-09:
I've found a similar issue. This issue is for the command `lora.power_mode(LoRa.TX_ONLY)`. Note that this is not initialisation.

When the power_mode is TX_ONLY, I expect the radio to go to sleep when it is not transmitting. This is true after executing that command. But once you send something, the power mode appears to be LoRa.ALWAYS_ON.

This time I turned off WiFi to get more stable power readings:

```python
from network import WLAN

wlan = WLAN(mode=WLAN.STA)
wlan.deinit()
```

* With the LoRa radio power_mode set to SLEEP, I measure 36.2 mA draw on a 5V power supply.
* With power_mode set to ALWAYS_ON, I measure 44.8 mA.
* With power_mode set to TX_ONLY, I measure 36.2 mA after executing the command (the same as sleep mode, as expected). But after sending some data I measure 44.8 mA (the same as ALWAYS_ON mode, which is unexpected).
* _FYI, the current consumption while data is being transmitted is 64.0 mA._

campbellsinclair, 2018-07-09:
Another issue: with bandwidth=BW_125KHZ and sf=12, it appears that the radio does not successfully send any data (because my other LoPy does not receive the data -- assumption: the error is with the transmitting LoPy).

If I continuously transmit data, the current is 60 mA for approximately 7 seconds. It then drops to 43 mA (which is the same as power_mode=ALWAYS_ON), even while attempting to transmit data.

robert-hh, 2018-07-09:
Cannot confirm. I have sent & received data at SF12BW125 with two xxPy devices.

campbellsinclair, 2018-07-09:
@robert-hh Try setting the coding rate to CODING_4_8 and send data which is 156 bytes or greater. It will send the first time, but if I send again it doesn't work.

Note that the LoPy v1 uses the SX1272 instead of the SX1276.

robert-hh, 2018-07-10:
There is a maximum airtime, depending on the transmission parameters. At SF12BW125 (DR=0), this is 51 (or 59) bytes. Setting the coding rate to 4/8 will reduce that further. So what were you test parameters?

nevercast, 2018-07-19:
Where is the maximum airtime defined, @robert-hh ? Do we have a millisecond hard limit for that?

nevercast, 2018-11-26:
The state machine certainly needs checking. The modem does not go to sleep when `power_mode` is set on the `LoRa` constructor. Can I help in any way? Some code to reproduce this issue?