pycom/pycom-micropython-sigfox #179: SIPY build return invalid PAC

syndycat, 2018-07-10:
Please include the following information when submitting a bug report:

SIPY 1.17.b3
1. We bought the SIPY one year ago... opened now for the first time
2. We tried update with pycome update tool  (1.15.1) ... but that failed on "checking firmware"
3. we make the build 1.17.b3 using make and bootloader (with SIPY board) 
4. we erased the flash 
5. we flashed the build sucesfuly 
6.  After flash calling pac() method is always returning b'\xff\xff\xff\xff\xff\xff\xff\xff'  like no mac is provisioned in the device

Xykon, 2018-07-10:
That is normal, the firmware updater registers the device and assigns the Sigfox credentials during the first upgrade.

syndycat, 2018-07-10:
It is not possible to do this from make procedure? We need this for continuous integration especially on L04 that will be planted on our PCB. If we have hundred pieces to flash we need automation.

Xykon, 2018-07-10:
We provide Sigfox credentials through the firmware updater for development purposes. If you wish to integrate Sigfox connectivity into your own product you will need to get your own Sigfox licenses. We can provide assistance with deploying these.