pycom/pycom-micropython-sigfox #180: Request; Feature; Storing Crypto state to NVRAM

nevercast, 2018-07-12:
* OEM LoPy
* Pycom MicroPython 1.19.0.b3 [v1.9.4-44b21d6] on 2018-07-01; LoPy with ESP32
```
(sysname='LoPy', nodename='LoPy', release='1.19.0.b3', version='v1.9.4-44b21d6 on 2018-07-01', machine='LoPy with ESP32', lorawan='1.0.2')
```

I'm developing sensors with the OEM LoPy, we've our first prototypes running and are very excited to finally have production ready boards after the R&D time spent, as for the firmware/software of our boards, we want our sensors to last a long time on battery in the field, we would also like to encrypt our LoRa traffic (We do not have a sufficient LoRaWAN infrastructure here in New Zealand yet). I have not tested my hypothesis yet but I assume that deep-sleep is going to kill the AES state (IV, Counter, etc), I would like a means of getting some of the AES state variables out of the crypto instance so that I can store them in NVRAM, then I can recreate my AES instance on wake.

My preference is having a function on the crypto instance along side decrypt and encrypt for .state() or .iv() or something similar. I would then use these values in the constructor of AES after waking up.

I'm happy to PR this work myself if we can agree on the implementation details, for another project that uses the WiPy I am maintaining many patch files that we apply to the firmware after each update from Pycom and I'd like to minimize my patch file work load by getting this functionality directly in to upstream.

Looking forward to discussing this feature.

Kind regards,
Josh.
Brush Technology.

iwahdan88, 2018-07-12:
Hi Josh,

It is not recommended to reuse IV data , For every message encrypted with the same key a different IV should be used as it is mentioned in pycom documentation.

> To avoid security issues, IV should always be a random number and should never be reused to encrypt two different messages. The same applies to the counter in CTR mode. You can use crypto.getrandbits() for this purpose. 

I am afraid we cannot add this functionality as it may introduce security issues.

nevercast, 2018-07-12:
The IV would not be the same in CFB and CBC modes because the specification for these modes updates the IV as an output parameter, so that these modes can be used in streaming functionality. I would like to still use the streaming functionality but across deep-sleep.

See the note in the ESP32 spec.
http://esp32.info/docs/esp_idf/html/d5/dbb/esp32_2include_2hwcrypto_2aes_8h.html#a5e7142edb96781127774244ae902551c

The IV will remain the same on both ends of the transaction in streaming mode since the new IV is generated based on the ciphertext, this means the IV needs to be sent for the first packet and not thereafter; this reduces 16 bytes from my LoRa payload which reduces my duty cycle of on time, and I believe it also aids my security by keeping my IV out of the public.

This would make it easier for me to move from CBC with an IV in every packet, to CFB. Which offers slightly better benefits against attacks.

I'm fairly certain this information I've presented is correct and secure. But if you know a security professional they can certainly weigh in on this.