pycom/pycom-micropython-sigfox #188: LTE cannot send AT commands if NB-IoT network gets unreachable

arapostol, 2018-07-26:
Pycom board: FiPY
Firmware version: 1.18.0

```
(sysname='FiPy', nodename='FiPy', release='1.18.0', version='v1.8.6-849-046b350 on
2018-06-01', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')
```

After attaching and connecting to my internal NB-IoT network (Amarisoft)...

```
from network import LTE

>>> lte = LTE()
>>> lte.attach(band=20)
>>> lte.isattached()
True
>>> lte.connect()
>>> lte.isconnected()
True
```

if the network is unavailable after a while (I manually stop the network)

```
>>> lte.send_at_cmd('AT+CEREG?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: LTE modem is in data state, cannot send AT commands
>>> lte.isconnected()
False
>>> lte.isattached()
True
>>> lte.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: modem already connected
>>> lte.disconnect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: the requested operation failed
>>> lte.send_at_cmd('AT+CEREG?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: LTE modem is in data state, cannot send AT commands
```

I think the modem knows that it is not connected to the network, but the LTE object is not so sure about that: it is returning that it not connected but if I try to send an AT command or connect again it is returning that it is already connected!

johanneshanek, 2020-12-01:
i have the some problems

peter-pycom, 2020-12-01:
Are you sure you have the exact same problem? Specifically this sequence:

```
>>> lte.isconnected()
False
>>> lte.isattached()
True
>>> lte.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: modem already connected
```

I am not sure what the implementation was back in version 1.18.0, but in today's implementation, I would not imagine this can happen. 

What can happen is that if you first lte.connect() and then you try to send AT commands, you will get `OSError: LTE modem is in data state, cannot send AT commands`. Even if the modem loses connectivity, the modlte stack will not be able to detect this directly. It will still return lte.isconnected()=True and unable to send AT commands. you can a) wait and see whether the modem manages to reconnect, or b) send lte.disconnect() or lte.pppsuspend() and get the AT interface back. 

This is inconvenient, but it is by design. We can't really make this a whole lot more convenient. In the latest release 1.20.2.r2, we've adjusted the lte callback https://docs.pycom.io/tutorials/networks/lte/#lte-connectivity-loss this gives a little bit more control

johanneshanek, 2020-12-01:
Thank you for your fast answer.
I will make a new firmware update.

peter-pycom, 2020-12-01:
Sure. 

Actually speaking about reaction speed, @arapostol , sorry you didnt' get a response in time. I'm going to close this issue, since I don't think you can reproduce this same situation with recent firmware. If I am wrong and you actually are able to reproduce - do reopen!

closed 2020-12-01 by peter-pycom