pycom/pycom-micropython-sigfox #189: utime.ticks_diff: Arguments swapped in 1.19.0.b4

robert-hh, 2018-07-28:
In branch development, version 1.19.0.b4, the arguments of ticks_diff are swapped compared to the master branch. Very confusing!

husigeza, 2018-08-03:
Hello,

The swap is needed because 1.19.0.b4 supports the uasyncio library which expects the swapped arguments :)

robert-hh, 2018-08-05:
You can make that even order-agnostic by changing the respective line (73? 79?) in uasyncio/core.py from
```
                    delay = time.ticks_diff(t, tnow)
```
into
```
                    delay = time.ticks_diff(t, tnow) * time.ticks_diff(2, 1)
```

husigeza, 2018-08-07:
Hello,
The order-agnostic can be confusing. I would keep the swapped order, furthermore the official MicroPython version of ticks_diff() has the same layout, with (end, start) prototype.

robert-hh, 2018-08-07:
Hello Géza, I understand the reason, and it is good to keep the two major variants of MicroPython close, but it stays a change, which breaks an unknown codebase of your existing customers. ticks_diff() is not an exotic functions, so it may be use at various places. For me, it is not so much of an issue. I know (do i?) where I'm using it. But it will confuse people. So unless Pycom decides to use the micropython.org codebase for uasyncio, it looks for me better to keep the existing order of ticks_diff() and change uasyncio/core.py.

martijnthe, 2018-10-02:
+1 for making this consistent with the official micropython.
It's confusing right now.

In general, it's frustrating that PyCom's firmware is so much behind on the micropython upstream.
Running into bugs all the time that have already been fixed in the upstream...

closed 2019-10-07 by robert-hh