pycom/pycom-micropython-sigfox #193: Recover former LoRa MAC?

nevercast, 2018-08-07:
Please include the following information when submitting a bug report:

* LoPy L01
* 1.19.0.b3

```
(sysname='LoPy', nodename='LoPy', release='1.19.0.b3', version='v1.9.4-44b21d6 on 2018-07-01', machine='LoPy with ESP32', lorawan='1.0.2')
```

* Exact steps to cause this issue
    1. `esptool -X -c esp32 --tx-break -b 921600 --after no_reset -p COM3 erase_flash`
    2. `from network import LoRa`
    3. `lora = LoRa(mode=LoRa.LORA, region=LoRa.AU915)`
    4. `lora.mac()`
* What you expected
    * To see `aa-bb-cc-dd-lo-ra-ma-cc`
* What happened instead
    * "And its gone", `ff-ff-ff-ff-ff-ff-ff-ff`

I have discovered the hard way that the LoRa MAC is not stored in some fuses but in flash, I've deleted it. Is there anyway to recover the former LoRa MAC address of this device? If I provide you with the WiFi MAC do you have the LoRa MAC available? The unique id is `24-0a-c4-0a-ab-c0`

Cheers.
Josh.

robert-hh, 2018-08-07:
That's not an issue, and was discussed quite a few times in the forum. And in your list above, after step i the flash should be completely empty. Especially step ii is not possible without reloading a firmware image. You can restore the Lora MAC by performing a  regular firmware upload with the PyCom updater.

Xykon, 2018-08-07:
The software server will only know the LoRa mac of the device if it was updated with the Pycom firmware tool at least once before it was erased. The latest beta firmware updater has a function to assign a new Mac (pending availability) if the mac was erased without a backup being made. The download links for the beta firmware updater are available on the forum.

closed 2018-08-07 by Xykon

nevercast, 2018-08-07:
@robert-hh Aye, I missed the step where I reflashed the 1.19.0.b3 image.

Thanks, both, I'll try the Firmware Updater and see if that recovers the mac, if not, I guess I'll get a new one. Cheers.