pycom/pycom-micropython-sigfox #194: pin_sleep_wakup function is missing

Bigboy20769, 2018-08-11:
I want to make sure that I am correct about the firmware issue.  I am struggling tried to run a simple deep sleep function test, but finally, I think I found out the culprit.  The function pin_deepsleep_wakeup is not included in the LoPy4 firmware version 1.19.0.b2, instead it is pin_sleep_wakeup! (word deep is missing).  I am indeed puzzled why no one noticed it.

Orginal code not working:
machine.pin_sleep_deepwakeup(['P8'], mode=machine.WAKEUP_ANY_HIGH, enable_pull=True)
machine.deepsleep(40)

Modified code working:
machine.pin_sleep_wakeup(['P8'], mode=machine.WAKEUP_ANY_HIGH, enable_pull=True)
machine.deepsleep(40)

Dillip

robert-hh, 2018-08-11:
You're right. Same for 1.19.0.b4 vs 1.18.0.r1. The definition in esp32/mods/modmachine.c was changed, for whatever reason. Not the only breaking change in version 1.19.0, and not the first breaking change by PyCom. 
People might not have noticed because that's the development version.

Sympatron, 2018-08-21:
I don't know why it was changed, but I think it would be a good idea to introduce an alias.