pycom/pycom-micropython-sigfox #198: TX/RX only pymakr upgrade does not work with V1.9 firmware?

hallard, 2018-08-27:
I designed lot of custom boards to handle lopy's for customers. On each board to be able to update and get control, I added a classic FTDI 6 pins connector.

As you can see schematics below can't be simpler
[image]

Everything is working fine, when on atom + pymakr can upload code without any problem with safe-boot or not activated.

But this is working only for firmware 1.8 I tried several time to use with firmware 1.9 and I see my soft output on pymark console (the print I do on my code), but each time I'm trying an upload, it stuck, it reset the device then block, no file are updated or whatever and I need to reset the device with reset button.
I tried with Lopy and Lopy4 same effect.

Did I missed something ? It's really anoying because I need to stay in V1.8 since my devices are on field with customers. And not being able to update firmware remotely with them when doing tests is a real issue.

Here are some screenshot of the issue with V1.19
[image]
[image]

And with version V1.8 all is fine
[image]

Any idea of what's going wrong ?

robert-hh, 2018-08-27:
The difference I see is the line:
`Type "help()" for more information.`
Which is printed by 1.18.0 and not by 1.19.0. Some character level protocols wait for a specific message, until they start to upload. This is hard to add, since with safe no start code but the frozen _boot.py is executed.

Xykon, 2018-08-27:
Hello Robert,

I don't think that's the problem. I'm currently in China visiting our manufacturing site and the way I debug our production scripts is that I connect a 2nd FTDI module to the TX pin only... this allows me to see pretty much everything that's going on. We use FTDI chips in the Expansion Board 2 and I'm not aware of any issues using Pymakr with it. What FTDI chip are you using exactly... maybe I can do some testing tomorrow.

robert-hh, 2018-08-27:
@xykon I do not have this problem, @hallard brought that up.
I'm not using pymakr at all, thus I have no issues with it.

robert-hh, 2018-08-27:
But nevertheless, inside pymakr, in lib/board/pyboard.js, there is this code snippet:
```
  safe_boot(cb,timeout){
    var _this = this
    this.logger.info("Safe boot")
    this.send_wait_for(CTRL_F,'Type "help()" for more information.\r\n>>>',function(err){
      _this.logger.info("Safe boot done...")
      if(cb) cb(err)
    },timeout)

  }
```
Which to me looks like doing a Ctrl-F safe boot and waiting for this 'Type "help()" for more information.\r\n>>>' string. And that one is missing at 1.19.0.b4. It is conditioned out in the firmware by MICROPY_PY_BUILTINS_HELP being not set to 1.

Xykon, 2018-08-27:
Thanks for checking... I'll assign this to our lead firmware developer for review.

iwahdan88, 2018-08-28:
Hi @hallard , can I know which version of Pymkr you are using?

hallard, 2018-08-28:
@iwahdan88 I'm always using the latest one that atom suggest, so 1.4.1 f pymakr-atom and the problem always been there since V1.9 of Lopy firmware, I didn't pay attention at the beginning because I wasn't using it.
But the issue occurs with 1.4.1, 1.4.0 and also 1.3.5 and lower

robert-hh, 2018-08-28:
Adding 
#define MICROPY_PY_BUILTINS_HELP  (1)
to mpconfigboard.h activates the text 'Type "help()" for more information.\r\n>>>' , and then pymakr upload works with version 1.19.0.b4.