pycom/pycom-micropython-sigfox #201: Suspected bug: missing bounds check for pycom.ota_write

martijnthe, 2018-09-05:
See `updater_write` function here:
https://github.com/pycom/pycom-micropython-sigfox/blob/master/esp32/ftp/updater.c#L122

I suspect that when using the `pycom.ota_write`, it's possible to write beyond the OTA partition (and into the partitions after it, such as the filesystem one). I don't see any bounds checking in the `updater_write` function nor in the micropython binding code.

cc @Xykon

martijnthe, 2018-09-05:
There is also no checking whether `pycom.ota_start` has been called before `pycom.ota_write` gets called, which probably would attempt a write to flash address 0x0. Probably harmless, but I'd probably add a check to make sure the code doesn't even hit the flash driver at all in this scenario.

robert-hh, 2018-09-05:
You are right. These checks can easily be implemented.

martijnthe, 2018-09-05:
I think there's another bug in `updater_write` regarding to the `spi_flash_erase_sector` call to erase the next sector. Isn't there a possibility that this erases a sector after the partition? If `pycom.ota_write` gets called for the last chunk and it happens to cause `updater_data.current_chunk` to equal `SPI_FLASH_SEC_SIZE` then it would erase past the partition I think.

martijnthe, 2018-09-05:
I think I found yet another bug: what if `pycom.ota_write` is called with a buffer larger than the portion that is erased in `updater_write`?

robert-hh, 2018-09-05:
Please feel free to submit a PR