pycom/pycom-micropython-sigfox #203: Controling TxPower in LoRaWAN mode

Prafullapp, 2018-09-07:
I want to control the TxPower in LoRaWAN mode,so i have few queries:
1. How to control the TXPower in LoRaWAN mode or is it directly controlled by LoRaWAN MAC???
2.  Does the power control will overall affect the device power consumption???
Although,I know in raw Lora power can be controlled by lora.init(),but i am not finding anything for LoRaWAN. 
Thanks in advance for the help.

robert-hh, 2018-09-07:
1. Sorry, I do not know
2. According to the data sheet, the power used affects the total current consumption, which while sending is between 18 mA at +7dB and 125mA at +20 dB (Sx1272, LoPy and FiPy) or 20mA to 120mA  for the Sx1276 (LoPy4).
But questions like these are better raised at the forum: https://forum.pycom.io/

Prafullapp, 2018-09-07:
Thank you for the response,I will raise it in the pycom forum.