pycom/pycom-micropython-sigfox #205: Firmware creates and uses main.py/boot.py and not main.mpy/boot.mpy

martijnthe, 2018-09-11:
- Board: LoPy 4
- Firmware: v1.18.1.r1

Steps to reproduce:
- Use `mpy-cross` to create a `main.mpy` and `boot.mpy` and load them onto an empty device's filesystem into `/flash/main.mpy`.
- Reboot the device.

Expected results:
- I would expect `main.mpy` and `boot.mpy` to be executed, but they are not being executed.

Actual results:
- `main.py` and `boot.py` are being created upon boot and used instead of the `.mpy` files. Using `os.listdir()` one can see the `.py` files have been created, alongside the `.mpy` ones...

robert-hh, 2018-09-11:
That seems intentional and is hardcoded. You could try to add these lines to boot.py:
```
from machine import main
main("main.mpy")
```
Is then at least main.mpy executed? Obviously, you could give your file also a different name.

martijnthe, 2018-09-12:
As a work-around, I ended up having single `import main_alt` / `import boot_alt` lines in the `main.py`/`boot.py` and that would just import the `main_alt.mpy` and `boot_alt.mpy`.