pycom/pycom-micropython-sigfox #206: WIFI antenna selection not working

johandse, 2018-09-12:
I am testing WIFI signal strength with external antenna on WiPy 3 with latest FW (1.18.1.r1). I am using this antenna: https://pycom.io/product/external-wifi-antenna/

I scan for networks with this code:

from network import WLAN
wlan = WLAN(mode=WLAN.STA)
wlan.antenna(WLAN.EXT_ANT)
nets = wlan.scan()
for net in nets:
	print(net)

I usually see rssi-values in the range -65 - -70, the same when I use the internal antenna. When I put my thumb over the integrated antenna rssi goes to appr. -85. This makes me belive that this antenna is in use.

Adding this code makes the board behave as expected.

from machine import Pin
p_out = Pin('P12', mode=Pin.OUT)
p_out.value(1)

dmartauz, 2018-09-12:
I can confirm same behaviour with LoPy4 running FW 1.18.1.r1. Difference in RSSI in my case is more than 20dB because the module (and its integrated antenna) is enclosed in aluminium housing.

robert-hh, 2018-09-12:
Looks like the bug is in esp32/util/antenna.c. For the "new" boards, the pin is not initialized as output, and the command to select the antenna just writes the value to the output register,

Jollyjohn, 2019-01-13:
@sdaniel55 Hi Daniel - this bug was raised against 1.18.1.r1 and since then we have had r3, r4 and r7. Is the fix in production now, or is it waiting for a change in the lowest dot.level? (e.g. 1.18.2)

robert-hh, 2019-01-13:
There is code to init P12 as output (that was missing) both in the actual master branch and in the release-candidate branch. It is used when the external antenna is selected.

closed 2019-03-01 by iwahdan88