pycom/pycom-micropython-sigfox #211: firmware problem Sipy

AnasShalyan, 2018-09-19:
Hi 
I am using Sipy with excitation board Pytrack. I have 2 problems
1- when i am trying to erase the memory using the windows update firmware i got this error 
[image]
2- when i am trying to see my Id and Pac using Atom it shows me in a wrong way 
[image]
This problem could not make me able to send data to Sigfox and i think it is because it sends wrong Id number. I have tried to use Fipy and it works ones then i have the same problem 
Best regards

Xykon, 2018-09-19:
Have you tried with different USB cables? We generally recommend a high quality USB cable as short and thick as possible.

AnasShalyan, 2018-09-19:
@Xykon hi i just tried that it is the same problem

AnasShalyan, 2018-09-19:
from network import Sigfox
from pytrack import Pytrack
import urequests as requests
from L76GNSS import L76GNSS
import socket
import time
import pycom
import struct

py = Pytrack()
gps = L76GNSS(py, timeout=60)

init_timer = time.time()

print("connecting to Sigfox")
# init Sigfox for RCZ1 (Europe)
sigfox = Sigfox(mode=Sigfox.SIGFOX, rcz=Sigfox.RCZ1)

# create a Sigfox socket
s = socket.socket(socket.AF_SIGFOX, socket.SOCK_RAW)
# make the socket blocking
s.setblocking(True)

s.setsockopt(socket.SOL_SIGFOX, socket.SO_RX, False)

# Post an location to the Wia cloud via Sigfox backend
def post_location(latitude, longitude):
    try:
        print(str(latitude), ":", str(longitude))
        s.send(struct.pack('f',float(latitude)) + struct.pack('f',float(longitude)))
    except:
        pass

# main loop
while True:
    final_timer = time.time()
    diff = final_timer - init_timer
    # Get coordinates from pytrack
    coord = gps.coordinates()

    # If the GPS has coordinates and 15 minites has past. Post the location data
    if not coord == (None, None) and diff < 900:
        lat, lng = coord
        post_location(lat, lng)
        init_timer = time.time()
I am using this code to send GPS data to sigfox. I have tried the same code with Fipy and it works ones then it does not working with both Sipy and Fipy