pycom/pycom-micropython-sigfox #214: Build LoPy4 firmware -> Error: No rule to make target 'build/LOPY4/release/frozen_mpy/frozen/Base/_boot.mpy', needed by 'build/LOPY4/release/frozen_mpy.c'.

maximilianschwab, 2018-09-28:
I would like to build the firmware for my LoPy4 on my own.
The following steps are running successfully without any errors:
* `$ cd esp32`
* `$ make BOARD=LOPY4 clean`
* `$ make BOARD=LOPY4 TARGET=boot`

But when I call `$ make BOARD=LOPY4 TARGET=app` I get the following error:

`Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
python ../py/makeversionhdr.py build/LOPY4/release/genhdr/mpversion.h
make: *** No rule to make target 'build/LOPY4/release/frozen_mpy/frozen/Base/_boot.mpy', needed by 'build/LOPY4/release/frozen_mpy.c'.  Stop.`

I'm using commit b0520f1e17f395952fdf01065865fc71a07e596a and Windows 10.

Any help would be greatly appreciated!

Xykon, 2018-09-28:
Did you build mpy-cross?

Prior to building the main firmware, you need to build mpy-cross

$ cd mpy-cross && make clean && make && cd ..

maximilianschwab, 2018-09-28:
Hi @Xykon,

sorry, I did not mention that I also built mpy-cross successfull prior building the main firmware without any errors.

Xykon, 2018-09-28:
OK just making sure... we're mainly using Linux & MacOS for development & release so it's possible there is an issue building the firmware on Windows we're not currently aware of.

I can look into that early next week as I'm travelling back from China tomorrow.

maximilianschwab, 2018-09-28:
Okay, thank you very much!

emmanuel-florent, 2018-09-28:
I have just tried to replay the issue but it went fine. My bet is that you are trying to add python code ? I had the same issue you mentioned but it was when using the addition of my own frozen module. If that s it I can help you further, ...

closed 2018-09-28 by emmanuel-florent

cmisztur, 2018-10-24:
#220 solves for me on Windows 10