pycom/pycom-micropython-sigfox #215: Timeout for Sending AT Commands

innoagg, 2018-10-08:
1.18.1.r1

GPy, FiPy

LTE.send_at_cmd("")

Would be helpful to be able to set the timeout in micropython so we can execute AT commands that take longer than the 9500ms timeout that is baked into the firmware. There appears to be no work-around since artificially creating a longer wait will risk the AT responses being lost.  A real problem when trying to attach to specific towers or scan for towers.

robert-hh, 2018-10-08:
That's true. Some commands, like AT+CPOS=?, take very long to respond.

iwahdan88, 2018-11-30:
Hi @innoagg , this option is available now as per v1.18.1.r6, you can set the parameter `delay` to wait specific amount of time before checking for response from modem on top which the 9500ms timeout is exercisable ... I'll update that in coming releases to just have a configurable timeout.

closed 2018-11-30 by iwahdan88