pycom/pycom-micropython-sigfox #216: Pysense/Pytrack: using SD card after sleep

campbellsinclair, 2018-10-11:
WiPy3.0, release='1.18.1.r1', version='v1.8.6-849-b0520f1 on 2018-08-29'

Tested on both:
Pytrack v1.1 with pytrack_0.0.8.dfu firmware
Pysense (v1.0 and v1.1) with pysense_0.0.8.dfu firmware

SanDisk Ultra 32GB micro SD HC I

* Exact steps to cause this issue
```
import pysense
pysense_board = pysense.Pysense()
pysense_board.setup_sleep(10)
pysense_board.go_to_sleep(False)

# wait for reboot

import machine
sd = machine.SD()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: the requested operation failed
```

I expected to be able to access the SD card after sleeping. If I eject and then reinsert the SD card after sleeping, it works.

campbellsinclair, 2018-11-28:
Hi @iwahdan88,

Tested with Lexar 32GB microSDHC and this card doesn't have the issue.

Tested again with Sandisk Ultra 32 GB microSDHC I and the issue definitely exists for that card.

iwahdan88, 2018-11-30:
Ok thanks @campbellsinclair , I will close the issue for now as it seems a problem with this particular SD card

closed 2018-11-30 by iwahdan88