pycom/pycom-micropython-sigfox #217: AS923 Class C

kingbite, 2018-10-15:
Please include the following information when submitting a bug report:

* Lopy
* Pycom MicroPython 1.18.1.r1 [v1.8.6-849-b0520f1]
* I tested Class C LoRa with EU868 first, this worked like expected. When trying Class C on AS923, no data is received at the LoPy. During these tests I used the proper HW and server settings.

Joining works like expected but the data sent by the server is never 'seen' by the LoPy

emmanuel-florent, 2018-10-15:
Hi @kingbite here @emmanuel-florent from Pycom, I have to make a setup for AS923 Class C me too these days so I can help to figure what is the problem. This might be in the way to use add_channel and remove_channel, what happens after OTAA when you uplink ? nothing ? one packet out of 8 ? 
Meanwhile, what gateway are you using ? what server ?

kingbite, 2018-10-16:
Im not sure what you mean I should do with the add_channel methods. Uplink (to the gateway) worked normally. When I put my gateway in class A and left my LoPy in class C, it could receive after the uplink. (so it was practically doing class A at that moment)

kingbite, 2018-10-16:
I am using a CloudGate from Option. http://www.option.com/

emmanuel-florent, 2018-10-16:
Well 

```
FiPy 1.18.1.r1 LoRaWAN 1.0.2
[1970-1-1 0:0:4] Starting LoRa Node with device eui 70b3d5499583e692
Adding channel up 1 923.2 Mhz
...
[1970-1-1 0:0:10] OTAA took 6 secs.
[1970-1-1 0:0:10] Using AS923 channel 1 encoding SF7BW125, DR 5
[1970-1-1 0:0:15] tx_time_on_air: 52 ms @dr 5I just made my setup here is some cut/paste of code:
```

```
self.lora = LoRa(mode=LoRa.LORAWAN, region=region, device_class=LoRa.CLASS_C, adr=False)
self.lora_otaa_join(self.lora)
# create a LoRa socket
self.lora_socket = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
# self.lora_socket.setsockopt(socket.SOL_LORA, socket.SO_DR, self.config["lora"]["data_rate"])
self.lora_socket.setsockopt(socket.SOL_LORA, socket.SO_DR, self.config["lora"]["data_rate"])

# msg are confirmed at the FMS level
self.lora_socket.setsockopt(socket.SOL_LORA, socket.SO_CONFIRMED, 0)

# make the socket non blocking y default
self.lora_socket.setblocking(False)

self.lora.callback(trigger=( LoRa.RX_PACKET_EVENT |
                        LoRa.TX_PACKET_EVENT |
                        LoRa.TX_FAILED_EVENT  ), handler=lora_cb)

time.sleep(4) # this timer is important and caused me some trouble ...

# In order for the network-server to be able to send data to the device,
# the device must first send an uplink frame so that the network-server
# knows which gateway is within reach of your device.
for i in range(0, 3):
    self._send_ntp_query()
    time.sleep(6)
```

```
def prepare_channels(lora, channel, data_rate):
    if not channel in range(1, 9):
        raise RuntimeError("channels should be in 1-8 for AS923")

    for i in range(0, 8):
        lora.remove_channel(i)

    upstream = (item for item in AS923_FREQUENCIES if item["chan"] == channel).__next__()

    # set default channels frequency
    lora.add_channel(int(upstream.get('chan')), frequency=int(upstream.get('fq')), dr_min=0, dr_max=data_rate)

    from lora_regions import human_fq

    print("Adding channel up %s %s" % (upstream.get('chan'), human_fq(upstream.get('fq'))))
    return lo
```
```
def lora_otaa_join(self, lora):
    # create an OTA authentication params
    dev_eui = binascii.unhexlify(self.config["lora"]["otaa"]['dev_eui'])
    app_key = binascii.unhexlify(self.config["lora"]["otaa"]['app_key'])
    nwk_key = binascii.unhexlify(self.config["lora"]["otaa"]['nwk_key'])

    start_join_time = time.time()

    # lora.nvram_restore()

    if not lora.has_joined():
        if self.config["lora"]["region"] == "US915":
            lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_key, nwk_key), timeout=0, dr=0)
        if self.config["lora"]["region"] == "AS923":
            lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_key, nwk_key), timeout=0, dr=2)
        else:
            lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_key, nwk_key), timeout=0, dr=int(self.config["lora"]["data_rate"]))

    # wait until the module has joined the network
    while not lora.has_joined():
        time.sleep(1)
        pycom.rgbled(0x000000)
        print('.', end='')
        time.sleep(1)
        pycom.rgbled(0x0000ff)

    lora.nvram_save()

    end_join_time = time.time()
    duration = (end_join_time - start_join_time)
    if duration > 0:
        print("")
        self._log("OTAA took %s secs." % duration)
    pycom.rgbled(0x00ff00)
```

kingbite, 2018-10-17:
Hi Emmanuel, 

First of all, thanks for looking at this in the first place. I'm still not able to get this working. How do I use the prepare_channels method?

this is my code: (uploaded to the wrong folder, I know)

`https://github.com/kingbite/lora_c_as923/blob/master/.gitignore/LoRa.py` 

When I get this working I will clean everything up and write comments so people can benefit from this example. 
Thanks in advance

kingbite, 2018-10-18:
Also, with this code I am able to send uplink but **not downlink** Please don't just let this pass. Are you able to receive?

kingbite, 2018-10-23:
Update: I received sample code from Pycom and will perform tests tomorrow. I will post an updated status after this with the working code.

kingbite, 2018-11-06:
@albertodicagno 

The sample code still does not seem to work for me for receiving messages in class C. I am waiting for support.

albertodicagno, 2018-11-06:
@kingbite i am experiencing problems with loran class C / class A confirmed uplink: I only get confirmation for the first uplink, starting from the second I receive no confirmation. I am using Brocaar's loraserver, and in the logs I can see that from the second uplink it says something like unmarshal error (it should be protocol buffer error in Go): margin out of range, range is -32. Any ideas? I am struggling with this problem. Pycom docs and sample code seem to be very lacking in general.

kingbite, 2018-11-06:
That is not really related to this issue. Please mail pycom and/or create a seperate ticket. I'm not familiar with Brocaar's loraserver sorry.

sdaniel55, 2018-11-06:
Hi @albertodicagno 👋 

You could ask for help on our [forum](https//forum.pycom.io) where our community might be able to help you. This repo is for issues related to the firmware, so we try to keep general usage questions on the forum. Alternatively, you can contact us using the [form on our website](https://pycom.io/support-community/support/contact-support/).

Thanks @kingbite for the moderation 😉

albertodicagno, 2018-11-06:
@sdaniel55 Thank you for the clarification. Anyway, I think that this question can be related to a firmware issue (composition of the phypayload); moreover I wrote an email to Pycom support without getting any answer. We are frustrated since we bought more than 50 Pycom boards and, while experiencing issues, receiving no support; on the other hand, as stated in many forum posts/github issues, the docs are lacking at many points. I will open a dedicated issue, hoping to get some feedback.