pycom/pycom-micropython-sigfox #218: WLAN init/deinit leaking memory -- esp_wifi_deinit() not getting called

martijnthe, 2018-10-17:
* board: LoPy 4
* firmware version: `(sysname='LoPy', nodename='LoPy', release='1.18.1.r1', version='f5d0c68 on 2018-09-06', machine='LoPy with ESP32', lorawan='1.0.2')` -- it's a custom build, but very close to `1.18.1.r1`.

I'm using `WLAN().deinit()` to (attempt) to shut down the wifi stack.
All is well, but after a bunch of on/off toggling, I'm seeing error messages these (I've turned on error logging in the firmware):
```
W (11106275) wifi: alloc eb len=204 type=2 fail

W (11106275) wifi: m f probe rsp l=177

W (11106277) wifi: alloc eb len=204 type=2 fail

W (11106277) wifi: m f probe rsp l=177

W (11179758) wifi: alloc eb len=204 type=2 fail

W (11179758) wifi: m f probe rsp l=177

W (11179779) wifi: alloc eb len=204 type=2 fail

W (11179779) wifi: m f probe rsp l=177
```

It looks like the allocator that the wifi stack uses can no longer allocate blocks.
I tried searching for where these messages are coming from, but I'm afraid it's coming from within one of the ESP32 binaries/static libraries.

Inspecting the `modwlan.c`, I noticed `esp_wifi_deinit() is not getting called. According to the header:
```
/**
  * @brief  Deinit WiFi
  *         Free all resource allocated in esp_wifi_init and stop WiFi task
  *
  * @attention 1. This API should be called if you want to remove WiFi driver from the system
  *
  * @return ESP_OK: succeed
  */
esp_err_t esp_wifi_deinit(void);
```

I'm guessing that's the bug...?

martijnthe, 2018-10-17:
I narrow the issue down to `_thread` locks leaking memory:
https://github.com/pycom/pycom-micropython-sigfox/issues/219

iwahdan88, 2018-10-25:
This Bug is already solved in Development release , `esp_wifi_deinit()` is now called when you do `WLAN.deinit()`

I'll close the issue as it is already fixed in Dev

Thanks

closed 2018-10-25 by iwahdan88

martijnthe, 2018-10-25:
@iwahdan88 thanks for the response.
Re. Development release: when is the v1.19.x release going out of beta?