pycom/pycom-micropython-sigfox #219: _thread lock not being free'd properly?

martijnthe, 2018-10-17:
* board: LoPy 4
* firmware: `(sysname='LoPy', nodename='LoPy', release='1.18.1.r1', version='f5d0c68 on 2018-09-06', machine='LoPy with ESP32', lorawan='1.0.2')` -- custom build but very close to `1.18.1.r1`.

It seems like locks that are created using `_thread.allocate_lock()` do not get free'd properly (or perhaps I don't understand micropyton's memory management model?).

See below for a little experiment that will end up triggering an `MemoryError: can't create lock` exception. From that point on the wifi stack also stops working, see https://github.com/pycom/pycom-micropython-sigfox/issues/218

According to `machine.info()` there's plenty of free heap space, but I guess the locks are allocated on a different heap (it would be nice to be able to show stats about that heap as well...)

How can I get rid of these leaky locks?

```
>>> import gc
>>> from _thread import allocate_lock
>>> for idx in range(0, 2000):
...     l = allocate_lock()
...     del l
...     if idx % 10 == 0:
...         gc.collect()
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: can't create lock

>>> import machine
>>> machine.info()
---------------------------------------------
System memory info (in bytes)
---------------------------------------------
MPTask stack water mark: 3564
ServersTask stack water mark: 1448
LoRaTask stack water mark: 1700
TimerTask stack water mark: 1940
IdleTask stack water mark: 576
System free heap: 258160
---------------------------------------------
```

iwahdan88, 2018-10-25:
Hi @martijnthe , I acknowledge the issue you pointed out, the problem was that the memory allocated for the lock creation was not taken from GC memory pool rather it was allocated from independent esp32 internal RAM, where GC had no control over. that's why `gc_collect` had no effect in the code above.
Also the reason you see the free heap size in response of `machine.info` not affected is that because the displayed heap is of a different Heap space that had no memory allocated for locks.

closed 2018-11-28 by iwahdan88

Sympatron, 2018-12-05:
Is this resolved in the development branch?

iwahdan88, 2018-12-05:
It is resolved in new release candidate that will be released this week.