pycom/pycom-micropython-sigfox #224: BLE Write Status Code is Not Returned

ansgarstrother, 2018-10-29:
From the documentation: 

characteristic.write(value)
Writes the given value on the characteristic. For now it only accepts bytes object representing the value to be written.

characteristic.write(b'x0f')

We would like the behavior to be: 

result_code = characteristic.write(value)

This is critical to our application since we return specific error codes that are critical to our application.

iwahdan88, 2018-11-28:
Hello @ansgarstrother thanks for your feedback, I do agree with you , I already created an internal ticket for modifying the BLE module to expose Error codes

closed 2018-11-28 by iwahdan88

ansgarstrother, 2019-06-14:
@iwahdan88 Was this feature implemented? We are circling back to this project and this feature is critical to our implementation.