pycom/pycom-micropython-sigfox #225: LTE modem becomes unresponsive - even after machine.reset()

stanleypa, 2018-11-02:
* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
G01
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
Pycom MicroPython 1.18.1.r2 [v1.8.6-849-3ab2277] on 2018-09-09; GPy with ESP32

I'm using the latest sequans modem firmware: - NB1-38729.dup

The issue happens sporadically maybe 1 in 50 or 1 in 100 times when trying to get a network connection. 

It seems to happen when sending and AT command -  'AT+COPS=1,2,"27201". This will often respond with "ERROR" and my code will try again. It will sometimes take quite a while to respond at all (the send_at_cmd method will timeout and return an empty string). In this case also I retry. Most of the time it eventually responds - the send_at_cmd returns the output from multiple AT+COPS commands and the connection is made. However sometimes this doesn't happen even after 5 minutes. At this point a watchdog kicks in to try to recover via a machine.reset() but the code then hangs on reboot at LTE object creation (lte=LTE()) 

Only a hard power reset will recover the system - not an option for normal deployed battery backed up systems! A machine.reset() should completely reset the modem also.

dmartauz, 2018-11-04:
I am experiencing the same issue with FiPy running Pycom FW 1.18.1.r2 and Sequans FW NB1-38729.

dmartauz, 2018-11-06:
@stanleypa 
Did you find any workaround for this issue in the meantime?
Just to be sure - is your G01 powered properly? I am thinking of possibility that Monarch chip cannot perform brownout reset correctly. However in our case the FiPy is powered from 2A power adapter and these hangs happen from time to time. I am thinking to connect oscilloscope to 3.3V pin of FiPy and find out what are minimal voltage levels during LTE transmissions.