pycom/pycom-micropython-sigfox #228: SQN3330 Upgrade

diganta2017, 2018-11-29:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy): **FIPY**
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
(sysname='FiPy', nodename='FiPy', release='1.18.1.r4', version='v1.8.6-849-dfa1176 on 2018-11-12', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')
```

I had previously ( in May 2018) upgraded my other Fipy with using Fipy_****33988.dup. But, on trying the same I am getting error:
>>> sqnsupgrade.run('/sd/FIPY_GPY_CATM1_33988.dup', 921600)
<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>
>>> FiPy with firmware version 1.18.1.r4
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sqnsupgrade.py", line 824, in run
  File "sqnsupgrade.py", line 174, in check_files
  File "sqnsupgrade.py", line 148, in __check_file
TypeError: 'int' object is not subscriptable 

  The "sqnsupgrade.py"file has only 153 lines, so how come error is beyond line 154.
   Also, I saw post on "https://software.pycom.io/downloads/sequans.html" and Github but I cn't seem to get authorization to download the zip files to proceed.
   Please advise as we have a over dozen Fipy's ready to be deployed for field testing. 
Thank you!

diganta2017, 2018-11-30:
>>> import os
>>> from machine import SD
>>> sd=SD()
>>> os.mount(sd,'/sd')
>>> os.listdir('/sd')
['upgdiff_38638-to-39529.dup', 'CATM1-39529.dup', 'upgdiff_33080-to-39529.dup', 'updater.elf']
>>> import sqnsupgrade
>>> sqnsupgrade.run('/sd/CATM1-38638.dup', '/sd/updater.elf')
<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>
>>> FiPy with firmware version 1.18.1.r4
Exception when checking file /sd/CATM1-38638.dup... wrong file name?
[Errno 2] ENOENT
>>> sqnsupgrade.run('/sd/CATM1-39529.dup', '/sd/updater.elf')
<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>
>>> FiPy with firmware version 1.18.1.r4
Attempting AT wakeup...
Session opened: version 1, max transfer 8192 bytes
Sending 55498 bytes: [########################################] 100%
Received ERROR from AT+SMSWBOOT=0,1! Aborting!
Unable to upgrade bootrom.
Unable to load updater from /sd/updater.elf
>>>

Xykon, 2018-11-30:
Hello,

With the new Sequans firmware updater please use the differential firmware upgrade files as explained here: https://forum.pycom.io/topic/3968/new-lte-firmware-release-v1-18-1-r3-cat-m1-firmware-modem-updater/26

The first time upgrade should use this file:

```
import sqnsupgrade
sqnsupgrade.run('upgdiff_33080-to-39529.dup')
```

or if using the SD card:

```
import sqnsupgrade
sqnsupgrade.run('/sd/upgdiff_33080-to-39529.dup')
```

Trying to use the old files like FIPY_GPY_CATM1_33988.dup can cause the modem to brick which is why we have recalled them quite a while ago.

diganta2017, 2018-11-30:
Hi Christian,

   Here is the result per your suggestion:

>>> os.mkfs('/flash')

>>> import os

>>> from machine import SD

>>> sd=SD()

>>> os.mount(sd,'/sd')

>>> import sqnsupgrade

>>> sqnsupgrade.run('/sd/upgdiff_33080-to-39529.dup')

<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>

>>> FiPy with firmware version 1.18.1.r4

Attempting AT wakeup...

Session opened: version 1, max transfer 8192 bytes

Sending 1211459 bytes: [########################################] 100%

Waiting for modem to finish the update...

This might take several minutes!

<<<=== DO NOT DISCONNECT POWER ===>>>

Upgrade failed with result failure[diffUpgrade,221] Patch didn't apply!

Please check your firmware file(s)

>>> 

 

But, when I run the ‘ATI’ command I get this after the upgrade attempt:

>>> from network import LTE

>>> lte=LTE()

>>> lte=LTE(carrier="verizon")

>>> lte.send_at_cmd('ATI')

'\r\nSEQUANS Communications\r\nVZM20Q\r\n'

>>> 

Does it mean it took it cause I see “SEQUANS “ or I should try the CATM1-39529.dup file. Would hate to try 39529 and brick the unit if I am wrong.

Best regards, 
Dig  

---------------------------------------------
Diganta D. Adhikari, MSCS,MSIT, PhD
Director of Engineering
IRROMETER Company, Inc. 
1425 Palmyrita Ave. 
Riverside, CA 92507 
(951) 682 9505
(951) 682 9501 Fax 
(951) 333 6075 Cell 
 <http://www.irrometer.com/> http://www.irrometer.com
DigA@irrometer.com  

Skype ID: diganta.adhikari

 

 <http://www.irrometer.com/> 

 

 <http://www.facebook.com/irrometer>        <https://twitter.com/Irrometer>        <https://www.youtube.com/user/irrometer?feature=mhum>

Xykon, 2018-12-01:
Can you please run the `sqnsupgrade.info(True)` command to check the detailed firmware versions of the different components. 

The good news is that your modem is still working and appears to be in application mode. 

I'm flying from China to Amsterdam today but we could connect through Skype tomorrow or Monday for further troubleshooting. Let me know what time/timezone would work for you and I'll try my best to accommodate you.

diganta2017, 2018-12-01:
Hi Christian,

   Thanks much for your prompt response. Have a safe trip back. Here is the Fipy’s response to :

>>> sqnsupgrade.info(True)

<<< Welcome to the SQN3330 firmware updater [1.2.0] >>>

>>> FiPy with firmware version 1.18.1.r4

Trying to detect modem state [1/5]

Your modem is in application mode. Here is the current version:

SYSTEM VERSION

==============

  FIRMWARE VERSION

    Bootloader0  : 5.1.1.0 [33080]

    Bootloader1* : BUILD5.1.1.0-33988 by robot-soft at 2017-12-11 15:06:33

    Bootloader2  : 5.1.1.0 [39529]

    NV Info      : 1.1,0,0

    Software     : 5.1.1.0 [33988] by robot-soft at 2017-12-11 15:04:15

    UE           : 5.0.0.0c

  COMPONENTS

    ZSP0         : 1.0.99-13028

    ZSP1         : 1.0.98-12295

LTE autostart enabled.

>>> 

    My schedule is open Monday 9:00-11:00 Am Mountain time in US, then I am off to the airport and won’t be back in my office until Thursday.

 

Best regards, 
Dig  

---------------------------------------------
Diganta D. Adhikari, MSCS,MSIT, PhD
Director of Engineering
IRROMETER Company, Inc. 
1425 Palmyrita Ave. 
Riverside, CA 92507 
(951) 682 9505
(951) 682 9501 Fax 
(951) 333 6075 Cell 
 <http://www.irrometer.com/> http://www.irrometer.com
DigA@irrometer.com  

Skype ID: diganta.adhikari

 

 <http://www.irrometer.com/> 

 

 <http://www.facebook.com/irrometer>        <https://twitter.com/Irrometer>        <https://www.youtube.com/user/irrometer?feature=mhum>

Xykon, 2018-12-03:
I'm available now if you can contact my on Skype (christian.ehlers.fr) or email (cehlers@pycom.io)

diganta2017, 2018-12-03:
Hi Christian,

   I apologize for the delayed response, with the upcoming travel later today, my day has been really crazy so far. Could we connect Thursday or Friday this week ( when I am back in the office). Also, did you get my last email regarding sqnsupgrade.info(True) ?

Thanking you!

Best regards, 
Dig  

---------------------------------------------
Diganta D. Adhikari, MSCS,MSIT, PhD
Director of Engineering
IRROMETER Company, Inc. 
1425 Palmyrita Ave. 
Riverside, CA 92507 
(951) 682 9505
(951) 682 9501 Fax 
(951) 333 6075 Cell 
 <http://www.irrometer.com/> http://www.irrometer.com
DigA@irrometer.com  

Skype ID: diganta.adhikari

 

 <http://www.irrometer.com/> 

 

 <http://www.facebook.com/irrometer>        <https://twitter.com/Irrometer>        <https://www.youtube.com/user/irrometer?feature=mhum>

iwahdan88, 2019-03-01:
Issue will be closed .. feel free to reopen if problem still exist

closed 2019-03-01 by iwahdan88