pycom/pycom-micropython-sigfox #232: utimeq overflows after reaching 0x3fffffff, while ticks_ms values will grows larger

martijnthe, 2018-12-05:
`utimeq` is mostly used in conjunction with `ticks_ms()` in [uasyncio], where "timestamps" generated by `ticks_ms()` are passed into the queue to sort the elements in the queue.

What the problem is depends a bit on what you deem "correct". The values returned from `ticks_ms()` can grow larger than `utimeq`'s fields to hold the time values. `peektime()` can only return values up to `0x3fffffff` (`(1 << 30) - 1`).

If one attempts to use [uasyncio] on a Pycom board, sooner or later (after 12.42 days when the overflow happens) mayhem will happen because of this wrap-around...

I guess one fix would be to make `ticks_ms()` wrap at `0x3fffffff` as well and fix `ticks_diff()` to return serial distance (taking wrap-arounds into account) -- see #231. Another could be to change `utimeq` to be able to handle larger time values.

Related issues: #231, #112.

cc @robert-hh @iwahdan88 

[uasyncio]: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L78

martijnthe, 2018-12-06:
Part of the problem: https://github.com/micropython/micropython/issues/4336

robert-hh, 2018-12-11:
See alsomy comment here: https://github.com/pycom/pycom-micropython-sigfox/issues/231

closed 2019-03-01 by iwahdan88