pycom/pycom-micropython-sigfox #243: NB-IoT firmware issues connecting GPy to Orange BE network

rma-ciss, 2019-01-04:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
import os
os.uname()
```
The output will look similar to this:
```
(sysname='LoPy', nodename='LoPy', release='0.9.4.b1', version='bc42ea7 on 2016-11-14', machine='LoPy with ESP32')
```
* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!

Xykon, 2019-01-04:
Hello Bart,

The newer NB-IoT firmwares use the same API as for CAT-M1. Please see https://docs.pycom.io/tutorials/lte/nb-iot.html

Regards, Christian

rma-ciss, 2019-01-07:
Christian,

OK, thx. After flashing the latest version of the FW on the GPy (1.18.1.r7) and with the modem FW NB1-40343 , it works. The GPy connects smoothly to the BEL Orange network!

Bart

closed 2019-01-07 by rma-ciss