pycom/pycom-micropython-sigfox #246: Pin: Setting the alternate function on a Pin causes a core dump

robert-hh, 2019-01-05:
Trying to set the alternate function on a Pin causes a core dump. Dump below:
```
Pycom MicroPython 1.20.0.rc3 [f0a4886-dirty] on 2019-01-03; LoPy4 with ESP32
Type "help()" for more information.
>>> from machine import Pin
>>> p=Pin('P11', Pin.IN, alt=1)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Core 1 register dump:
PC      : 0x40101f45  PS      : 0x00060b30  A0      : 0x800fa287  A1      : 0x3ffdfa60  
A2      : 0x3ffdfb00  A3      : 0x3f815dc0  A4      : 0x3ffdfb14  A5      : 0x00000000  
A6      : 0xffffff30  A7      : 0x3f815dc1  A8      : 0x80101e56  A9      : 0x3ffdfa40  
A10     : 0x3ffc2174  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x3ffdfb18  
A14     : 0x3ffdfa88  A15     : 0x3ffdfaf0  SAR     : 0x00000007  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4009cc68  LEND    : 0x4009cc73  LCOUNT  : 0x00000000  

Backtrace: 0x40101f45:0x3ffdfa60 0x400fa284:0x3ffdfb00 0x400f69e1:0x3ffdfb70 0x400f6a0e:0x3ffdfb90 0x400df158:0x3ffdfbb0 0x400df3a4:0x3ffdfc50 0x400de111:0x3ffdfc90

================= CORE DUMP START =================
nDQAABAAAABsAQAA
lFr8P6D5/T9o/v0/
UPr9PwD+/T9Q6AIASFf8P0hX/D+UWvw/QFf8PxQAAADkuv0/5Lr9P5Ra/D8AAAAA
BQAAAGzO/T9NaWNyb1B5AAAAAAAAAAAAAQAAAGj+/T8AAAAAIAkGAAUAAAABAAAA
AAAAAOSr/D8AAAAAAAAAAAAAAACIt/0/8Lf9P1i4/T8AAAAAAAAAAAEAAAAAAAAA
V2xAPwAAAAD80glAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAA==
AQAAAEUfEEAwCwYAh6IPgGD6/T8A+/0/wF2BPxT7/T8AAAAAMP///8FdgT9WHhCA
QPr9P3Qh/D8CAAAAAQAAABj7/T+I+v0/8Pr9PwcAAAAAAAAAAAAAAGjMCUBzzAlA
AAAAAAEAAAAY+/0/jDYIQPi8+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAEAAAAY+/0/iPr9P/D6/T/kaQ+AAPv9Pzis/D84rPw/
sPv9P6D6/T+Hog+AYPr9PwgAAAAo+/0/AAAAAAD7/T8k+/0/BQAAAAAAAAABAAAA
/////9D6/T/kaQ+AAPv9Pzis/D84rPw/cREQgH3NCUAIAAAAHPv9PwAAAAAFAAAA
AAAAAAD7/T8g+/0/KPv9PyT7/T8rAgAAKwAAAFcAAAAU+/0/AAAAADD////BXYE/
EWoPgHD7/T+wXIE/AAAAALBcgT/BXYE/EPv9Pxz7/T84rPw/dCH8P7YBAAADAAAA
fgcAAAMAAAD/////VwAAAPhBCIBg+/0/gPv9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
APv9PwAAAAAFAAAAQPv9P1vxDYCQ+/0/sFyBPwAAAAAAAAAAAAAAAAUAAAAAAAAA
p/MNgLD7/T+wXIE/FgAAAAAAAAAAAAAAAAAAAAIAAAAU4Q2AUPz9P/wiQD8AAAAA
AAAAAOD7/T+n8w2AsPv9P6xj/D8AAAAAFgAAACcAAAAiAAAA/v///w0AAACMY/w/
DQAAAAAAAAAU4Q2AUPz9P/wiQD8AAAAA6PANgJRggT8wYIE/MPz9P1D8/T/AIkA/
AAAAAIxj/D8bAAAAAgAAABYAAABQ/P0/YPgVAAAAAAAWAAAAJwAAACIAAAD+////
AAAAAJD8/T+g/P0/9Br8PyAAAAAbAAAAcFmBPwgAAAAAAAAAkPz9P6D8/T8AAAAA
SHv7PwAAAAABAAAAAAAAAAAAAAAA/v0/AAAAAAAAAAABAAAApaWlpaWlpaWlpaWl
AP//////////////////////////////////////////////////////////////
//////////////////////+lpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWl
paWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWl
paWlpaWlpaUCAAAAzSBAP9ggQD8DAAAABgAAAAAAAAABAAIAAAAAAAAAAAAAAAAA
AAAAAAEAAAAAAAAA/////wAAAAD/////////////////////xUITgQD/////////
/////////////////////////////////wAAADIAAAACAQAAAAAAAADgGwCQ/P0/
AAAoAAAAAAABAAAAuP39P2D9/T8AAAAAAAAAAAAAAAAAAAAAAQAAAExc/D/kq/w/
AAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAADP79PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
WKv9P+Cp/T9Qq/0/
4Kn9P/Cq/T/MdlUY1LD9P+RW/D9Yq/0/3Fb8PxkAAADiWUmrBGHclVir/T8AAAAA
AAAAAFSn/T9JRExFAAj08zu3IoFt1ugAAAAAAFCr/T8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACIt/0/8Lf9P1i4/T8AAAAAAAAAAAEAAAAAAAAA
V2xAPwAAAAD80glAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGoSXw==
6DYIQN7UH0AwAQYA5FgJgKCq/T8gAAAAAAAAAEz6/D8MGfw/0Fj8P3hW/D9o+vw/
gKr9PwAAAABQDwAAAAAAAMB3/j8DAAAAIwAGAAoAAACT0g6AYKr9P0iUDkCHlA5A
AAAAAOE4CEDAd/4/TGcJQOhp+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADAd/4/AwAAACMABgAAAAAAwKr9PwAAAAAAAAAA
TPr8PwwZ/D/QWPw/eFb8PwAAAADwqv0/AAAAAAAAAAAMGfw/AAAAAAAAAAAAAAAA
AQAAAAAAAAAEAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPyq/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
[... 235 lines cut ...]
AAAAACXoAgBUNfw/YDb8P/////8Q1vs/AAAAMDDZ+z8AAAAAAAAAAAAAAAABAAAA
AAAAADDW+z9UNfw/YDb8P/////+EmxNA9QAAAAQAAAAAAAAAcNb7PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAADYXP0/GQAAAA0AAAAOAQAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAfNb7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
4J79P+A1/j8EN/4/
4DX+P6A2/j+lpaWlDKX9P2BW/D/gnv0/WFb8PxQAAAClpaWlpaWlpeCe/T8AAAAA
BQAAAAgr/j9Tb2NrZXQgT3BlcmF0aW8AAQAAAAQ3/j8AAAAApaWlpQUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACIt/0/8Lf9P1i4/T8AAAAAAAAAAAEAAAAAAAAA
V2xAPwAAAAD80glAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWlpQ==
6DYIQMA4DkAwAAUAAAAAAKA2/j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAJj1+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAACsNv4/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
================= CORE DUMP END =================
Rebooting...
```

Xykon, 2019-01-06:
Hello Robert,

I think the function argument is misleading... please refer to the ESP32 [Technical Reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf).

The alt argument here expects the signal number (see Table 17 in the TRM).

It seems that if you assign a signal to a pin, such as U0RXD_in/out (14) it is unassigned from the original GPIO pin, for example you loose the REPL if you try p=Pin('P11', Pin.IN, alt=14)
I believe the reason it crashes when passing 1 (signal SPIQ_in/out) is that the processor no longer has access to the flash and SRAM because the quad spi IO signal is being re-routed to a new pin.

We should probably add some code to not allow such "dangerous" IO re-assignments. And check the documentation to make sure it properly explains what this argument does. Thanks a lot for bringing it up.

robert-hh, 2019-01-06:
OK. That was just a blind trial, not considering the TRM. I'll close that for now. But at least it should be documented.

closed 2019-01-06 by robert-hh