pycom/pycom-micropython-sigfox #252: v1.20-rc6: partition layout incompatible with old bootloaders?

martijnthe, 2019-01-25:
@iwahdan88 I noticed you changed the flash partitioning in 3eb11ef3ea30922aa5383ae8bf8ed4ec6d20e723, increasing the size of the factory + ota slots by ~200KB or so.

I guess this means you cannot update to v1.20 if you have an older bootloader, because the image size etc. is hard-coded in the bootloader (i.e. `IMG_SIZE` etc.)?

Is my assumption correct?

What new part(s) made the firmware grow so much?

martijnthe, 2019-01-25:
@iwahdan88 / @Xykon would you take a PR that adds a `#define` to make it possible to leave out openthread? It takes up a whopping 125KB of code space:

[image]
(Generated graph with https://github.com/armmbed/mbed-os-linker-report, swapped out `arm-none-eabi-nm` with `xtensa-esp32-elf-nm`)

iwahdan88, 2019-01-25:
Hello @martijnthe , you are correct regarding the OTA update from stable to New 1.20 Firmware, currently this is not possible because there is no way Now to update the partition table over the air, we are currently researching this matter for possible solutions.

As for the PR, yes of course,  please submit it and we will review and included in next release. Thanks very much for your help.

martijnthe, 2019-01-25:
@iwahdan88 there you go:
https://github.com/pycom/pycom-micropython-sigfox/pull/253

closed 2019-03-01 by iwahdan88