pycom/pycom-micropython-sigfox #256: uart pin modification

davestones, 2019-02-02:
(sysname='FiPy', nodename='FiPy', release='1.18.1.r7', version='v1.8.6-849-d1c5ea9 on 2018-12-17', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1')

FiPy on Expansion3
P21 and P22 connected to provide UART loopback

[1] - uart 'constructor' doesn't accept pins as per online documentation, must use init
[2] - use of object after uart.deinit causes divide by zero panic

[1]
>>> uart = machine.UART(1, baudrate=9600, pins=('P21', 'P22'))
>>> uart.init(baudrate=9600, bits=8, parity=None, stop=1)
>>> uart.write('hello')
5
>>> time.sleep(0.5)
>>> uart.any()
0
# 0 bytes received, expected 5...

>>> uart = machine.UART(1, baudrate=9600)
>>> uart.init(baudrate=9600, bits=8, parity=None, stop=1, pins=('P21', 'P22'))
>>> uart.write('hello')
5
>>> time.sleep(0.5)
>>> uart.any()
5
# 5 bytes received as expected

[2]
read or write after deinit causes catastrophic failure
>>> uart.deinit()
>>> uart.any()
0
>>> uart.read(1)
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero)
. Exception was unhandled.
# Exception

davestones, 2019-02-02:
Doh… Thanks for the clarification - of my incompetence - I shouldn't be 'trying' to code and watch rugby at the same time...

robert-hh, 2019-02-02:
Don't worry. At least you pointed at a real bug. I just made a patch for that.

closed 2019-03-07 by iwahdan88