pycom/pycom-micropython-sigfox #258: wipy help('modules') not working

johncblacker, 2019-02-04:
Running WiPy 3.0, firmware:
>>> import os
>>> os.uname()
(sysname='WiPy', nodename='WiPy', release='1.18.1.r7', version='v1.8.6-849-d1c5ea9 on 2018-12-17', machine='WiPy 

When I try help('modules') I get a list of string functions.  Doesn't/isn't the WiPy support this command?  If not, what's the alternative?

jb
Thank you!

robert-hh, 2019-02-04:
Help modules was enabled in the release candidate version 1.20.0.rc5. Will come soon in the master branch or 1.20.x will get the master branch. You get this version by checking the box "Include development releases" in the pycom updater.

johncblacker, 2019-02-04:
Great, updated to the development firmware .....rc4 and help('modules') works!  Many thanks!

closed 2019-02-04 by johncblacker