pycom/pycom-micropython-sigfox #259: machine.sleep([time_ms], resume_wifi_ble) works just once

alvaroalvarezvazquez, 2019-02-04:
Please include the following information when submitting a bug report:

* Pycom board: WiPy 3.0
* firmware version: V1.20.0.RC6

* Exact steps to cause this issue
    1. I place the machine.sleep(10000, True) within my while loop.
For the first run, the function worked as expected, the wipy slept for 10 seconds, it woke up, automatically reconnected to the Wifi, do some stuff with UDP sockets, and went back to the beginning of the loop with a new machine.sleep(10000, True)
  2. Unfortunately for the 2nd run, it raised an Errno 118 EHOSTUNREACH.
I found that this error come from an incorrect net connection. 
  3. I solved it just placing a ConnectToWlan() routine just after waking up [ConnectToWlan() is my routine to perform wlan.init+ wlan,ifconfig+ wlan.connect... which was just intended for the start up, NOT for waking up from a lightsleep). But this is just to solve the issue through a dirty trick. 

* What you expected
- Since the  value for resume_wifi_ble was True always, it was expected to enables restoring after wakeup any WiFi or BLE connection that was interrupted by light sleep

* What happened instead
- The True value just enable restoring of the wifi JUST ONCE, this means, for the first run of the while loop. For the second and succesive runs it does not reconnect

Thank you!

Changed-Daily, 2019-02-13:
Please include the following information when submitting a bug report:
Pycom board: OEM module W01
firmware version: WiPy-V1.20.0.rc7
Exact steps to cause this issue
Initialise Bluetooth and start advertising
Call to machine.sleep(5000, True)
What you expected
After the 5000ms sleep period I expected the module to wake up and resume Bluetooth advertising.
What happened instead
I get error message:
ASSERT_PARAM(512 0), in rwble.c at line 230 and the module hangs, indefinitely. Any attempts at deinit() and init() before and after sleep result in the same error.

iwahdan88, 2019-02-20:
Hello @alvaroalvarezvazquez ,
Thanks for reporting this issue , a fix has been already applied and will be pushed to release-candidate branch shortly
Regards

Changed-Daily, 2019-02-20:
Hi, can you comment on the Bluetooth issue also please?

iwahdan88, 2019-02-21:
> Hi, can you comment on the Bluetooth issue also please?

This Bluetooth issue is related to an open issue on ESP-IDF 
https://github.com/espressif/esp-idf/issues/880

alvaroalvarezvazquez, 2019-02-21:
> Hello @alvaroalvarezvazquez ,
> Thanks for reporting this issue , a fix has been already applied and will be pushed to release-candidate branch shortly
> Regards

Thank you so much to the oycom team for aolving the bug!!

closed 2019-03-01 by iwahdan88