pycom/pycom-micropython-sigfox #263: V1.20.0.rc7: Build for Fipy fails with undefined reference

robert-hh, 2019-02-09:
Error message:
```
LINK build/FIPY/release/application.elf
build/FIPY/release/application.a(modlte.o):(.literal.TASK_LTE_UPGRADE+0xc): undefined reference to `xMeshTaskHndl'
```
xMeshTaskHndl is not defined in the sources and not in esp-idf, branch idf_v3.1

iwahdan88, 2019-02-09:
Hi @robert-hh , thanks for reporting this, you are correct
in modlte.c
line 319 should be conditioned like this
```
#ifdef LORA_OPENTHREAD_ENABLED
    vTaskSuspend(xMeshTaskHndl);
#endif
```
we will update it asap

robert-hh, 2019-02-09:
Tested & it works

Xykon, 2019-02-09:
I also created a PR for this https://github.com/pycom/pycom-micropython-sigfox/pull/264

closed 2019-02-09 by robert-hh

robert-hh, 2019-02-11:
Thanks. But If I was the only one noticing that  as a problem, then a separate branch would not have to be used. I made that change anyhow already before posting the issue. Now there is another branch in the repository, which will be abandoned soon.