pycom/pycom-micropython-sigfox #268: Unable to set static IP when WiFi is in AP mode

Milan991, 2019-02-19:
Pycom could not set static IP address when WiFI is in AP mode.
`wlan.ifconfig()` keep returning ('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0').

Code that I'm using:

`import pycom
from network import WLAN

wlan = WLAN()
print("Turning on wlan: " + ssidName)
wlan.init(mode=WLAN.AP, ssid=ssidName, auth=(WLAN.WPA2, password), hidden=False)
wlan.ifconfig(id=1, config=('192.168.4.0', '255.255.255.0', '192.168.4.1', '192.168.4.1'))`

Xykon, 2019-02-25:
You need to use `wlan.ifconfig(id=1)` to see the ifconfig information for access point mode.

See 
https://docs.pycom.io/firmwareapi/pycom/network/wlan.html#wlanifconfigid0-configdhcp-or-configtuple for more details.

Milan991, 2019-02-25:
@Xykon Great, thank you. Closing issue.

closed 2019-02-25 by Milan991