pycom/pycom-micropython-sigfox #269: Pysense Firmware Update errors out at 96% with bytesError completion packet

remohn, 2019-02-23:
I get an error when trying to update the Pysense firmware with pysense_0.0.8.dfu. It errors out at 96% completion.

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 04d8
Match product ID from file: f011
Opening DFU capable USB device...
ID 04d8:f011
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
WARNING: Runtime device already in DFU state ?!?
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 64
Copying data from PC to DFU device
Download        [======================== ]  96%        15808 bytesError sending completion packet

Thank you!

remohn, 2019-02-23:
Gave up on Windows 10 and update the board from Ubuntu. Much simpler.

closed 2019-02-23 by remohn

walook, 2020-04-24:
Hello, 

I have the same problem as remohn, I understand that it is simpler from Ubuntu, but let me reopen the question. Has anyone encountered this problem and found a solution?

Thanks !

cartovarc, 2020-07-07:
[image]

Same problem