pycom/pycom-micropython-sigfox #270: LoRa(mode=LoRa.LORAWAN, region = LoRa.US915) uses the wrong frequency table for the US

tcottle, 2019-02-25:
(sysname='LoPy', nodename='LoPy', release='1.18.2', version='v1.8.6-849-743b7e7 on 2019-01-21', machine='LoPy with ESP32', lorawan='1.0.2', pybytes='0.9.6')

Cannot join the TTN network using the default US channels in the Lopy.  

pycom/pycom-micropython-sigfox/blob/master/lib/lora/mac/region/RegionUS915.c set the channels from 902.3 to 903.7MHz

void RegionUS915InitDefaults( InitType_t type )
{
    switch( type )
    {
        case INIT_TYPE_INIT:
        {
            // Channels
            // 125 kHz channels
            for( uint8_t i = 0; i < US915_MAX_NB_CHANNELS - 8; i++ )
            {
                Channels[i].Frequency = 902300000 + i * 200000;
                Channels[i].DrRange.Value = ( DR_3 << 4 ) | DR_0;
                Channels[i].Band = 0;
}

The correct channelization is 903.9 to 905.3MHz and can be found at
https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html

Adding this code fragment after creating a lora object will allow the Lopy to reliably join the TTN network

lora = LoRa(mode=LoRa.LORAWAN, region = LoRa.US915)

for index in range(0, 7):
  lora.remove_channel(index)

for index in range(16, 72):
  lora.remove_channel(index)

lora.add_channel(8, frequency=903900000, dr_min=0, dr_max=3)
lora.add_channel(9, frequency=904100000, dr_min=0, dr_max=3)
lora.add_channel(10, frequency=904300000, dr_min=0, dr_max=3)
lora.add_channel(11, frequency=904500000, dr_min=0, dr_max=3)
lora.add_channel(12, frequency=904700000, dr_min=0, dr_max=3)
lora.add_channel(13, frequency=904900000, dr_min=0, dr_max=3)
lora.add_channel(14, frequency=905100000, dr_min=0, dr_max=3)
lora.add_channel(15, frequency=905300000, dr_min=0, dr_max=3)
lora.add_channel(65, frequency=904600000, dr_min=4, dr_max=4)  # 500KHz uplink larger dr breaks(?)

oligauc, 2019-03-02:
The Lorawan frequency band for the US region is from 902.3 to 914.9 for the 125KHz channels and from 903 MHz to 914 MHz for the 500KHz channels.

The default configuration is not targeting any specific lorawan server.

tcottle, 2019-03-02:
I understand.  A US pycom customer that has a TTN gateway will not be able to get the combination working following the published examples.  Perhaps at better solution would be to keep the LoRa.US915 definition as is but add a LoRa.USTTN definition as well. 

One other small issue.  There are methods for setting and clearing the channels but no method to see how the current channels are set