pycom/pycom-micropython-sigfox #273: v1.20.0.rc7: Enabling OrderedDict feature; fails to compile

brotherdust, 2019-02-28:
* Exact steps to cause this issue
    1. Added `#define MICROPY_PY_COLLECTIONS_ORDEREDDICT          (1)` after line 78 in 'esp32/myconfigport.h`
    2. Then, I started the usual compile process. Note that compile is successful with this feature disabled.
* What you expected
    * Successful compile
* What happened instead
    * Failed compile with output below:
```
make V=1
Use make SECURE=on [optionally SECURE_KEY ?= secure_boot_signing_key.pem] to enable Secure Boot and Flash Encryption mechanisms.
BASE Variant
python ../py/makeversionhdr.py build/WIPY/release/genhdr/mpversion.h
CC ../py/objdict.c
xtensa-esp32-elf-gcc -DESP_PLATFORM -Wpointer-arith -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wl,-EL -mlongcalls -Wall -ffunction-sections -fdata-sections -fno-common -Wno-sign-compare -Wno-old-style-declaration -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-unused-parameter -DWITH_POSIX  -mfix-esp32-psram-cache-issue -Os -nostdlib -std=gnu99 -g3 -ggdb -fstrict-volatile-bitfields -Iboards/WIPY -DLORA_OPENTHREAD_ENABLED -I. -I.. -Ihal -Iutil -Imods -Itelnet -Iftp -Ilora -Ilte -Ican -Ibootloader -Ifatfs/src/drivers -Ilittlefs -Ibuild/WIPY/release -Ibuild/WIPY/release/genhdr -I/home/jarad/pycom-esp-idf/components/bootloader_support/include -I/home/jarad/pycom-esp-idf/components/bootloader_support/include_priv -I/home/jarad/pycom-esp-idf/components/mbedtls/mbedtls/include -I/home/jarad/pycom-esp-idf/components/mbedtls/port/include -I/home/jarad/pycom-esp-idf/components/driver/include -I/home/jarad/pycom-esp-idf/components/driver/include/driver -I/home/jarad/pycom-esp-idf/components/heap/include -I/home/jarad/pycom-esp-idf/components/esp32 -I/home/jarad/pycom-esp-idf/components/esp32/include -I/home/jarad/pycom-esp-idf/components/esp_adc_cal/include -I/home/jarad/pycom-esp-idf/components/soc/include -I/home/jarad/pycom-esp-idf/components/soc/esp32/include -I/home/jarad/pycom-esp-idf/components/expat/include -I/home/jarad/pycom-esp-idf/components/freertos/include -I/home/jarad/pycom-esp-idf/components/json/include -I/home/jarad/pycom-esp-idf/components/expat/include -I/home/jarad/pycom-esp-idf/components/lwip/include/lwip -I/home/jarad/pycom-esp-idf/components/lwip/include/lwip/port -I/home/jarad/pycom-esp-idf/components/newlib/include -I/home/jarad/pycom-esp-idf/components/newlib/platform_include -I/home/jarad/pycom-esp-idf/components/nvs_flash/include -I/home/jarad/pycom-esp-idf/components/spi_flash/include -I/home/jarad/pycom-esp-idf/components/tcpip_adapter/include -I/home/jarad/pycom-esp-idf/components/log/include -I/home/jarad/pycom-esp-idf/components/sdmmc/include -I/home/jarad/pycom-esp-idf/components/vfs/include -I/home/jarad/pycom-esp-idf/components/bt/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/device/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/device/include/device -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/bta/dm -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/bta/hh -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/bta/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/bta/sys/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/common/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/stack/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/stack/gatt/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/stack/gap/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/stack/l2cap/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/btcore/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/osi/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/hci/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/gki/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/api/include -I/home/jarad/pycom-esp-idf/components/bt/bluedroid/btc/include -I../lib/mp-readline -I../lib/netutils -I../lib/oofatfs -I../lib -I../drivers/sx127x -I../ports/stm32 -I/home/jarad/pycom-esp-idf/components/openthread/src -DMICROPY_NLR_SETJMP=1 -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DESP_PLATFORM -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DREGION_AS923 -DREGION_AU915 -DREGION_EU868 -DREGION_US915 -DBASE=0 -DPYBYTES=1  -DVARIANT=0 -DNDEBUG -c -MD -o build/WIPY/release/py/objdict.o ../py/objdict.c
../py/objdict.c:567:13: error: 'MP_QSTR_OrderedDict' undeclared here (not in a function)
     .name = MP_QSTR_OrderedDict,
             ^
make: *** [../py/mkrules.mk:47: build/WIPY/release/py/objdict.o] Error 1
```

Thank you!

robert-hh, 2019-02-28:
You have to run:
make clean
first. Then the QSTR table will be rebuilt, and MP_QSTR_OrderedDict is known. That I a common pitfall when changing mpconfigport.h.

brotherdust, 2019-02-28:
You. Freaking. Rock.
Thank you!

closed 2019-02-28 by brotherdust