pycom/pycom-micropython-sigfox #284: Unable to transmit all messages LoraWAN TTN lopy4/pysense

morrison93, 2019-04-02:
Hey Guys!

I'm using two lopy4 each one attached to two pysense boards and i'm trying a simple nano-gate - node link over ttn, simple stuff with the code provided in https://docs.pycom.io/tutorials/lora/lorawan-nano-gateway.html

* The firmware version in each one is:
Node - (sysname='LoPy4', nodename='LoPy4', release='1.18.2.r3', version='v1.8.6-849-86b2375 on 2019-03-15', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

nanoGate - (sysname='LoPy4', nodename='LoPy4', release='1.18.2.r2', version='v1.8.6-849-5cf02dc on 2019-03-07', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

in the main.py of the node i'm running a cycle to send four equal strings strings with a time space of 1 minute between  messages as it is in the code below: 
""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""
for i in range (4):
    pycom.rgbled(0x007f00) # green
    time.sleep(60)
    pycom.rgbled(0x7f7f00) # yellow
    time.sleep(1)
    data = "AHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHH"
    s.send(data)
    print("enviou" + str(i) + str(rtc.now()) + str(data)) # printa numero + data
    time.sleep(3)
    pycom.rgbled(0x7f0000) # red

    rx = s.recv(256)
    if rx:
        print(rx)
    time.sleep(6)

pycom.heartbeat(True)
"""""""""""""""""""""""""""""""""""""""""""""""""
 
I would expect to see 4 uplinks in the ttn traffic annd 4 uplinks in the gateway but this doesn't happen. I get 1 messages sometimes 2 (printscreens below) although in  Atom Repl i get the 4 prints, can someone help me troubleshoot this please or give me a logical explanation?

________________________________________________
Q2: Can someone explain why i always keep getting a message from the nano-gateway of 157 bytes? is this some control message between the nano-gateway and the ttn?

Thank you!

[image]
[image]
[image]

robert-hh, 2019-04-02:
How did you register your Node at TTN, by OTAA or ABP?
Do you see at the gateway also only a subset of the messages?
One reason could be, that the node is configured e.g. by the OTAA response to use more than one frequency, but the nanogateway looks at a single frequency only. That's why the examples set all channels to that single frequency after the node has been registered.

morrison93, 2019-04-02:
The node is set by OTAA

robert-hh, 2019-04-02:
OTAA may transmit a list of frequencies to use in the response message.

I used these lines after the join:
```
# remove all channels but the first
for i in range(1, 16):
    lora.remove_channel(i)
lora.add_channel(0, frequency=config.LORA_FREQUENCY, dr_min=0, dr_max=5)
```

morrison93, 2019-04-02:
That code in the node right?

closed 2019-04-04 by morrison93