pycom/pycom-micropython-sigfox #285: GPy: No Band8 after update to 1.18.2.r4 or 1.20.0.rc9

BlackDuck888, 2019-04-02:
Today i did an upgrade for my GPy to the latest revision. 

After that i get for 
```
lte.attach(band=8, apn="iot.1nce.net")
ValueError: band 8 not supported by this board hardware!
```
```
Sysname: GPy
Release: 1.20.0.rc9
Version: v1.9.4-7eeaf3f on 2019-04-01
Machine: GPy with ESP32
Query LTE modem imei... 35434**********
Query SIM iccid... 89882***************
Sent to modem: 'ATI1', delay 1000 ms
UE6.0.0.0
LR6.0.0.0-41019
OK
Sent to modem: 'AT!="showver"', delay 1000 ms
SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0  : 5.1.1.0 [41065]
    Bootloader1  : 6.0.0.0 [40343]
    Bootloader2* : 6.0.0.0 [41019]
    NV Info      : 1.1,0,0
    Software     : 6.0.0.0 [41019] by robot-soft at 2018-12-13 21:41:15
    UE           : 6.0.0.0
  COMPONENTS
    ZSP0         : 1.0.99-3171
    ZSP1         : 1.0.98-3170
```

But show caps says band 8 is supported

```

iwahdan88, 2019-04-02:
Hi @BlackDuck888 , what is the hardware version of your GPy , you can find it at the bottom of the board?

BlackDuck888, 2019-04-02:
GPy V1.0r

iwahdan88, 2019-04-02:
Yes this version only supports Bands [3, 4, 12, 13, 20, 28]. if you want to use other Bands you have to Submit an [RMA request](https://pycom.io/support-community/support/rma/) for replacement of your board with a new one.

BlackDuck888, 2019-04-02:
I will, but why they get sold without this support? 
Which Hardwarerevision support this native, for next shopping? 
and last why i don't work with vodafone band 20 ?

iwahdan88, 2019-04-02:
> I will, but why they get sold without this support?
See this [post](https://forum.pycom.io/topic/3843/important-news-regarding-cat-m1-and-nb-iot-cellular-range) 

> Which Hardware revision support this native, for next shopping?
version 1.2 supports 17 bands with sequanse Firmware 41065(CATM1) and 41019(NB-IOT)

> and last why i don't work with vodafone band 20 ?
v1.0r supports Band 20 , you should not have problem connecting.

Regards

BlackDuck888, 2019-04-02:
I process the RMA first.

The hardware version should be part of the documentation, i spend near a weak is this without success. an other mail ticket is not anwered/open and show caps shows me that band 8 is supported. 

i had no success, with vodafone and band 20 and i am tring much to get it work.

closed 2019-04-02 by BlackDuck888

iwahdan88, 2019-04-12:
Thanks @BlackDuck888 and sorry for your inconvenience, docs will be updated Asap.