pycom/pycom-micropython-sigfox #288: Unable to transmit full message Lorawan TTN lopy4/pysense

morrison93, 2019-04-05:
Hi guys!

I'm using a lopy4 attached to a pysense and I'm trying to send a temperature value to the ttn via lorawan. I want the temperature value with a decimal case so i'm multiplying it by 10 and for a value, let's say 288 (28,8 ºC) I pass the value to bytes and append it to a bytearray a proceed to send it to the ttn. 

Now, the problem is to trasnmit 288d i need 2 bytes and 288d = 120h = 0000 0001 0010 0000 
but when i transmit it with s.send(bytearray) in the ttn I only get a Payload of 20H so... I'm only sending the 1st byte of information. Why's that?? 

I've previously sent strings with many more bytes and decoded them successfully and i can't do this with a simple value? Why the different behaviour? Can you help me troubleshoot this?

Thank you!

oligauc, 2019-04-05:
Please can you try:
.........

ba = bytearray()
pkt = struct.pack('H',288)
ba.extend(pkt)

s.send(ba)

morrison93, 2019-04-06:
That solved it.... 
my bad 
Thank you very much!

closed 2019-04-06 by morrison93