pycom/pycom-micropython-sigfox #289: Fail to upgrade GPy LTE Modem

ThLOCP, 2019-04-10:
Dear all,

My board infos: a GPy  + PScan

sysname='GPy', nodename='GPy', release='1.18.2.r4', version='v1.8.6-849-ba178ae on 2019-04-01', machine='GPy with ESP32

I am getting really sad about my first trials with pycom/gipy:

-  1st GPy board, all was ok (LTE Modem upgrades, getting started, etc.) but after a few days the tiny LTE antenna connector broke on the board : too light to last more than a few days...
- 2nd GPy  board : impossible to upgrade the Sequans modem. First board this update worked perfectly but here it sounds bricked. Here is attached my log of the command
sqnsupgrade.run('/sd/CATM1-39529.dup', '/sd/updater.elf', debug=True)
NOTE: I know that a newer revision is available, the CATM1-41065, I downloaded it and my first try was with the newest revision. The error has been the same one.

Please help me recover my gipy.  

Best regards,
Thierry

[Thierry-on-pycom-io-AttachedLog002.txt](https://github.com/pycom/pycom-micropython-sigfox/files/3063576/Thierry-on-pycom-io-AttachedLog002.txt)

iwahdan88, 2019-04-10:
Hello @ThLOCP 

Can you run the command `sqnsupgrade.info(debug=True)` and post the log output?

ThLOCP, 2019-04-10:
Dear iwahdan88,  

I thank you for your anwer. Here is the log you requested:

>>> import sqnsupgrade
>>> sqnsupgrade.info(debug=True)
<<< Welcome to the SQN3330 firmware updater [1.2.5] >>>
>>> GPy with firmware version 1.18.2.r4
The current delay is 1000
Response (+++ #1): None
Response (AT #1) b'\r\nERROR\r\n'
Response (AT #2) b'\r\nOK\r\n'
Response (AT+SMOD?) b'\r\n2\r\nOK\r\n'
Converting response: 2 to int...
Modem state: 2
Your modem is in application mode. Here is the current version:
mirror? False  recover? False  resume? True  direct? True  atneg_only? False bootrom? False load_fff? False
baudrate: 921600 target_baudrate: 921600
SYSTEM VERSION
==============
  FIRMWARE VERSION
    Bootloader0* : 5.1.1.0 [33080]
    Bootloader1  : NA
    Bootloader2  : NA
    NV Info      : 1.1,0,0
    Software     : 5.1.1.0 [33080] by robot-soft at 2017-10-27 07:49:11
    UE           : Unknown
  COMPONENTS
    ZSP0         : 1.0.98-12323
    ZSP1         : 1.0.98-12295
IMEI: 354347091896347

I will let you check my log before sending the `sqnsupgrade.run('/sd/CATM1-39529.dup')`. Let me know.

Best regards,
Thierry

iwahdan88, 2019-04-10:
Hi @ThLOCP , looking again to your 1st log, we realised that the message `Received ERROR from AT+SMSWBOOT=3,1! Aborting!` indicates that the modem File system is in Read-Only state (due to a bug in modem firmware) which can only be recovered if the modem is in recovery Mode.
So you should submit an [RMA request](https://pycom.io/support-community/support/rma/) referencing this issue in the Description, we will then replace that board for you.

closed 2019-04-10 by iwahdan88

frontierconnect, 2021-01-16:
In case of OEMs that we soldered on our boards do you have an work around ?

robert-hh, 2021-01-16:
If you have P4, P8, P23, GND and 3v3 accessible, you could connect a small SD card breakout, like the one below. DI to P4, DO to P8 and SCK to P23, GNDto GND, 3V3 to Vss

[image]