pycom/pycom-micropython-sigfox #290: lte_check_attached returning true only if "AT+CEREG?" starts with "+CEREG: 2,1," or "+CEREG: 2,5,"

riveccia, 2019-04-10:
We are using a GPy board, with the latest development firmware, version "1.20.0.rc9".

Inside the modlte.c file, the method "lte_check_attached" is returning false when "legacy" parameter is true and modem is set to a CEREG value equal to 3, 4 or 5 (as per Monarch documentation):

```
if (((pos = strstr(modlte_rsp.data, "+CEREG: 2,1,")) || (pos = strstr(modlte_rsp.data, "+CEREG: 2,5,"))) && (strlen(pos) >= 31) && (pos[30] == '7' || pos[30] == '9')) {
   attached = true;
}
```

When we set CREG=2, "lte_check_attached" is returning true; when we set CREG=3 or 4, "lte_check_attached" is returning false even if the modem is attached to the network.

Thank you.

iwahdan88, 2019-04-10:
@riveccia yes, the `is attached()` method only supports the CEREG format 2 , if you want to use another one (3,4 or 5) you should check state of attachment yourself using [AT+CEREG?] and [AT+CGATT?]

riveccia, 2019-04-10:
is there any possibility the support will be included in a future release?

iwahdan88, 2019-04-10:
@riveccia , I think your use case here is that you also need to read the extra params offered by the other CEREG values, right? like `[,[<cause_type>],[<re
ject_cause>][,[<Active-Time>],[<Periodic-TAU>]]]]`

riveccia, 2019-04-11:
Yes, we would like to have access to that data.
We tried to monitor attachment state using AT commands, but the modem raises an exception when trying to connect to the network if CEREG is not equal to 2.