pycom/pycom-micropython-sigfox #294: GPy LTE CAT M1 and NB-IoT issue with hologram SIM in US

nvellal, 2019-04-18:
The following was tested in Northeast suburbs of Boston, MA, USA

GPy 1.0 with Pytrack 

Sequans firmware version
[image]

Output from
```import os  os.uname()```

The output is:
[image]

Ran the script attached below. The program line lte.attach() never returns true in the 
    while (True): lte.isattached() 
loop

[main_py.txt](https://github.com/pycom/pycom-micropython-sigfox/files/3094933/main_py.txt)

Thank you!
NV

iwahdan88, 2019-04-18:
@nvellal Please make sure the Network you are trying to connect to is on one of these bands [3,4,12,13,20,28] as that is what only supported by the GpY hardware v1.0

nvellal, 2019-04-18:
@iwahdan88 we were informed by Hologram that for NB-IoT it used T-Mobile and the bands are 2,4 and 12

iwahdan88, 2019-04-18:
@nvellal I can see that the LTE modem firmware you have is [40165] which is a CAT-M1 firmware for NB-IoT use [41019]

nvellal, 2019-04-18:
@iwahdan88 we have both SIMs (LTE and NB-IoT) and we tried both options [two different GPys with respective firmwares for CAT-M1 and NB-IoT] and neither of them worked. It looks like the modem is not responding, are there any diagnostic scripts to validate the status of the modem? How can we find out the problem is not with the connectivity?

mcopela, 2019-07-03:
Any luck fixing this issue? I've been working with GPy and a Hologram LTE-M sim as well, no success connecting.