pycom/pycom-micropython-sigfox #295: Cannot send Sigfox message when OneWire bus is being used

conorf50, 2019-04-20:
I'm developing an application that uses a DS18B20 temp sensor and sends the data to Sigfox every hour.

When trying to send a Sigfox message with the OneWire bus enabled, the board hangs and does not return to the REPL. A reset is required to make it work again.

If the last line is commented out. the code will print the current temp and return to the prompt. Likewise if the OneWire code is commented out, the Sigfox message will send. But if I try to use both at the same time it does not work.

I'm using the [DS18B20 example](https://github.com/pycom/pycom-libraries/tree/master/examples/DS18X20) as a base, I have not changed the OneWire.py or boot.py files at all.

Firmware Version - output of os.uname():
```
(sysname='SiPy', nodename='SiPy', release='1.18.2.r5', version='v1.8.6-849-396d71b on 2019-04-16', machine='SiPy with ESP32', sigfox='1.0.1')
```

My main.py:

``` python
from network import Sigfox # sigfox libs
import time 
from machine import Pin # listings for device pins
from onewire import DS18X20 # driver for the temperature sensor
from onewire import OneWire # onewire driver for DS18B20 driver above
import socket # sigfox
import struct # compression of float values

# init Sigfox for RCZ1 (Europe)
sigfox = Sigfox(mode=Sigfox.SIGFOX, rcz=Sigfox.RCZ1)
# create a Sigfox socket
s = socket.socket(socket.AF_SIGFOX, socket.SOCK_RAW)
# make the socket blocking
s.setblocking(True)
# configure it as uplink only
s.setsockopt(socket.SOL_SIGFOX, socket.SO_RX, False)

#DS18B20 temp sensor data line (yellow wire) connected to pin P8 (G15 on Expansion Board)
ow = OneWire(Pin('P8')) # enabling the onewire bus causes the device to freeze when sending sigfox message

temp = DS18X20(ow) 
temp.start_conversion()
time.sleep(1)
temp = round(temp.read_temp_async(), 2) # round the temp value to two decimal places
print("Temp =")
print(float(temp))

print("sending data")
#s.send(struct.pack('f',float(34.1)) + bytes([12])) # will freeze the board unless OneWire code is commented out!
```

conorf50, 2019-04-21:
I thought it was a power supply issue so I powered the SiPy with an external 5v 1 amp source. The issue still occurs.

conorf50, 2019-04-21:
Downgrading to [1.9.0.b1](https://software.pycom.io/downloads/SiPy-1.9.0.b1.tar.gz) stops the board from freezing.

When I have time, I will test newer versions.

husigeza, 2019-08-10:
Could you please check whether the issue exist on the latest official version: "1.18.2.r7" ?

conorf50, 2019-08-10:
@husigeza  Sorry, I no longer have a dev board to test with.