pycom/pycom-micropython-sigfox #297: occasional core dump on i2c.writeto()

gretel, 2019-04-29:
```
(sysname='WiPy', nodename='WiPy', release='1.20.0.rc10', version='v1.9.4-350eac8 on 2019-04-16', machine='WiPy with ESP32')
```
sometimes, for a reason unknown to me this call crashes and dumps core:
```
self._i2c.writeto(self._addr, framebytes)
```
when it happened the `wipy` needs to be disconnected from power to not crash all the time on this call. afterwards, it works fine, until it fails again.

```
***ERROR*** A stack overflow in task MicroPy has been detected.
abort() was called at PC 0x40097758 on core 1

Backtrace: 0x400975e7:0x3ffd7330 0x4009773f:0x3ffd7350 0x40097758:0x3ffd7370 0x40094187:0x3ffd7390 0x40095afc:0x3ffd73c0 0x40095ab2:0x00000000

================= CORE DUMP START =================
RGMAABEAAABsAQAA
/Ff8P3By/T/Mo/0/
wHP9P2Cj/T8lxgAArFT8P3hC/T/8V/w/pFT8PxQAAACU+v0/lPr9P/xX/D8AAAAA
BQAAANBz/T9NaWNyb1B5AAAAAAAAAAAAAQAAAMyj/T8AAAAAIQAGAAUAAAABAAAA
WGL7P5Sp/D8AAAAAAAAAAAkAAACIXP0/8Fz9P1hd/T8AAAAAAAAAAAEAAAAAAAAA
y3dAPwAAAAAUxQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAA==
MTU3N+d1CUAzAAYAQncJgDBz/T8vAAAALwAAAAwAAAD/////AAAAAP7///8AAAAA
EHP9PwAAAADBcv0/b3L9PzEAAAAAAAAAcnL9PwQAAAAdAAAAAAAAAIC+CUCLvglA
AAAAAG9y/T8xAAAAZDYIQEj6+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAwXL9PwAAAABbdwmAUHP9P7M6/D8AAAAA
AAAAAFBz/T9QWV9OVk0AAIpBCYBwc/0//Ff8PzRY/D9rTAmAoBP8PyAAAAAgAAAA
/1oJQJBz/T80Vvw/IwAFAKoyCYDQE/w/GBT8PwEAAAC1WglAwHP9PxgU/D8AAAAA
cHP9P7Bz/T9waHkAAAAAADhW/D+AU/w/wHP9P9Bz/T+Qc/0/AAAAAGNhY2hlX2Fk
yDYIQPC/AEAwDQYAzaYQgIB0/T8AAAAAIw0GACANBgDQyP0/AwAAACMABgBHhQCA
YHT9P0MBAAAAQAAA//8/s6urAAD+AAAAMHX9PwIAAADAdP0/AAAAAIC+CUCLvglA
AAAAAMg4CECrqwAAMFoJQEj6+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADIOAhAq6sAADBaCUBVphCAkHT9P6yt/D94rfw/
0qYQgLB0/T+srfw/eK38P///P7OrqwAA/gAAAAAAAABVphCA0HT9P6yt/D94rfw/
IAkGANDI/T8DAAAAIwAGANKmEIDwdP0/rK38P3it/D///z+zq6sAAP4AAACwdf0/
VaYQgBB1/T+srfw/eK38PyAFBgDQyP0/AwAAACMABgDSphCAMHX9P6yt/D94rfw/
//8/s6urAAD+AAAAAAAAAFWmEIBQdf0/rK38P3it/D8gAQYA0Mj9PwMAAAAjAAYA
0qYQgHB1/T+srfw/eK38P///P7OrqwAA/gAAADB2/T9VphCAkHX9P6yt/D94rfw/
IA0GANDI/T8DAAAAIwAGANKmEICwdf0/rK38P3it/D///z+zq6sAAP4AAAAAAAAA
VaYQgNB1/T+srfw/eK38PyAJBgDQyP0/AwAAACMABgDSphCA8HX9P6yt/D94rfw/
//8/s6urAAD+AAAAsHb9P1WmEIAQdv0/rK38P3it/D8gBQYA0Mj9PwMAAAAjAAYA
0qYQgDB2/T+srfw/eK38P///P7OrqwAA/gAAAAAAAABVphCAUHb9P6yt/D94rfw/
IAEGANDI/T8DAAAAIwAGANKmEIBwdv0/rK38P3it/D///z+zq6sAAP4AAAAwd/0/
VaYQgJB2/T+srfw/eK38PyANBgDQyP0/AwAAACMABgDSphCAsHb9P6yt/D94rfw/
//8/s6urAAD+AAAAAAAAAFWmEIDQdv0/rK38P3it/D8gCQYA0Mj9PwMAAAAjAAYA
0qYQgPB2/T+srfw/eK38P///P7OrqwAA/gAAALB3/T9VphCAEHf9P6yt/D94rfw/
IAUGANDI/T8DAAAAIwAGANKmEIAwd/0/rK38P3it/D///z+zq6sAAP4AAAAAAAAA
VaYQgFB3/T+srfw/eK38PyABBgDQyP0/AwAAACMABgDSphCAcHf9P6yt/D94rfw/
//8/s6urAAD+AAAAMHj9P1WmEICQd/0/rK38P3it/D8gDQYA0Mj9PwMAAAAjAAYA
0qYQgLB3/T+srfw/eK38P///P7OrqwAA/gAAAAAAAABVphCA0Hf9P6yt/D94rfw/
IAkGANDI/T8DAAAAIwAGANKmEIDwd/0/rK38P3it/D///z+zq6sAAP4AAACweP0/
VaYQgBB4/T+srfw/eK38PyAFBgDQyP0/AwAAACMABgDSphCAMHj9P6yt/D94rfw/
//8/s6urAAD+AAAAAAAAAFWmEIBQeP0/rK38P3it/D8gAQYA0Mj9PwMAAAAjAAYA
0qYQgHB4/T+srfw/eK38P///P7OrqwAA/gAAADB5/T9VphCAkHj9P6yt/D94rfw/
IA0GANDI/T8DAAAAIwAGANKmEICweP0/rK38P3it/D///z+zq6sAAP4AAAAAAAAA
VaYQgNB4/T+srfw/eK38PyAJBgDQyP0/AwAAACMABgDSphCA8Hj9P6yt/D94rfw/
//8/s6urAAD+AAAAsHn9P1WmEIAQef0/rK38P3it/D8gBQYA0Mj9PwMAAAAjAAYA
0qYQgDB5/T+srfw/eK38P///P7OrqwAA/gAAAAAAAABVphCAUHn9P6yt/D94rfw/
IAEGANDI/T8DAAAAIwAGANKmEIBwef0/rK38P3it/D///z+zq6sAAP4AAAAwev0/
VaYQgJB5/T+srfw/eK38PyANBgDQyP0/AwAAACMABgDSphCAsHn9P6yt/D94rfw/
//8/s6urAAD+AAAAAAAAAFWmEIDQef0/rK38P3it/D8gCQYA0Mj9PwMAAAAjAAYA
0qYQgPB5/T+srfw/eK38P///P7OrqwAA/gAAALB6/T9VphCAEHr9P6yt/D94rfw/
IAUGANDI/T8DAAAAIwAGANKmEIAwev0/rK38P3it/D///z+zq6sAAP4AAAAAAAAA
VaYQgFB6/T+srfw/eK38PyABBgDQyP0/AwAAACMABgDSphCAcHr9P6yt/D94rfw/
//8/s6urAAD+AAAAMHv9P1WmEICQev0/rK38P3it/D8gDQYA0Mj9PwMAAAAjAAYA
0qYQgLB6/T+srfw/eK38P///P7OrqwAA/gAAAAAAAABVphCA0Hr9P6yt/D94rfw/
[... 485 lines cut ...]
AAAAADDW+z84Mvw/RDP8P/////+k6xJA9QAAAIAHAAAAAAAAcNb7PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAADYAf0/GQAAAA0AAAAOAQAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAfNb7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
4EP9P8DW/T/s1/0/
wNb9P4DX/T+lpaWlBDP9P8RT/D/gQ/0/vFP8PxQAAAClpaWlpaWlpeBD/T8AAAAA
BQAAAPDL/T9Tb2NrZXQgT3BlcmF0aW8AAQAAAOzX/T8AAAAApaWlpQUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACIXP0/8Fz9P1hd/T8AAAAAAAAAAAEAAAAAAAAA
y3dAPwAAAAAUxQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKWlpQ==
yDYIQJw1DkAwAAUAAAAAAIDX/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAJgH+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAACM1/0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
================= CORE DUMP END =================
Rebooting...
ets Jun  8 2016 00:22:57
```

gretel, 2019-05-08:
does anybody copy?

iwahdan88, 2019-05-09:
Hello @gretel , This is clearly a stack overflow problem, is it possible to share the code that reproduces the issue?

gretel, 2019-05-17:
@iwahdan88 sure

```
from machine import I2C
i2c = I2C(2, mode=I2C.MASTER, pins=('P4', 'P8'), baudrate=20000)
i2c_scan = i2c.scan()
```