pycom/pycom-micropython-sigfox #301: lora.add_channel settings are ignored

choeffer, 2019-05-15:
Please include the following information when submitting a bug report:

* The Pycom board you are using: LoPy4
* The firmware version you are using. You can get it by executing the following Python code at the REPL:
os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.18.2.r6', version='v1.8.6-849-443304d on 2019-05-06', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

* Exact steps to cause this issue
    1. Take the LoRa Example from https://docs.pycom.io/tutorials/lora/lorawan-abp.html
    2. Then add _lora.add_channel()_ three times to overwrite the three default channels in EU868 (as mentioned in the wiki) to enable the LoPy4 to talk to a single channel gateway.

* What you expected
    * e.g. I expected the LoPy4 just to use this single channel
* What happened instead
    * e.g. Instead, I got that the LoPy4 was still sending LoRaWAN packages on all three default channels, see attached SDR waterfall.

If you need more information please let me know.

choeffer, 2019-05-15:
```
from network import LoRa
import socket
import ubinascii
import struct

# Initialise LoRa in LORAWAN mode.
# Please pick the region that matches where you are using the device:
# Asia = LoRa.AS923
# Australia = LoRa.AU915
# Europe = LoRa.EU868
# United States = LoRa.US915
lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868)

# create an ABP authentication params
dev_addr = struct.unpack(">l", ubinascii.unhexlify('XXX'))[0]
nwk_swkey = ubinascii.unhexlify('XXX')
app_swkey = ubinascii.unhexlify('XXX')

# join a network using ABP (Activation By Personalization)
lora.join(activation=LoRa.ABP, auth=(dev_addr, nwk_swkey, app_swkey))

# overwrite the default three channels to just sent on one channel
lora.add_channel(0,frequency=868100000, dr_min=5, dr_max=5)
lora.add_channel(1,frequency=868100000, dr_min=5, dr_max=5)
lora.add_channel(2,frequency=868100000, dr_min=5, dr_max=5)

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 5)

# make the socket blocking
# (waits for the data to be sent and for the 2 receive windows to expire)
s.setblocking(True)

# send some data
s.send(bytes([0x01, 0x02, 0x03]))

# make the socket non-blocking
# (because if there's no data received it will block forever...)
s.setblocking(False)

# get any data received (if any...)
data = s.recv(64)
print(data)
```

choeffer, 2019-05-16:
I did another test with the following code.

```
from network import LoRa
import socket
import ubinascii
import struct

# Initialise LoRa in LORAWAN mode.
# Please pick the region that matches where you are using the device:
# Asia = LoRa.AS923
# Australia = LoRa.AU915
# Europe = LoRa.EU868
# United States = LoRa.US915
lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868)

# create an ABP authentication params
dev_addr = struct.unpack(">l", ubinascii.unhexlify('26011730'))[0]
nwk_swkey = ubinascii.unhexlify('4AAD4C1CDD48D4086C57F618281A198E')
app_swkey = ubinascii.unhexlify('ECA9E5C1E4BBC10C33346A2526C56824')

# join a network using ABP (Activation By Personalization)
lora.join(activation=LoRa.ABP, auth=(dev_addr, nwk_swkey, app_swkey))

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# overwrite the default three channels to just sent on one channel
lora.add_channel(0,frequency=868100000, dr_min=5, dr_max=5)
lora.add_channel(1,frequency=868100000, dr_min=5, dr_max=5)
lora.add_channel(2,frequency=868100000, dr_min=5, dr_max=5)
lora.add_channel(3,frequency=867500000, dr_min=5, dr_max=5)

# send some data
s.send(bytes([0x01, 0x02, 0x03]))
```

In this case, a new channel is added and used, see attached screenshot.

[image]

So adding a fourth channel seems to work, but I am not able to overwrite the default channels. Also removing them and then add them again does not work.

What else could I test to provide more information?

choeffer, 2019-05-16:
In the https://github.com/pycom/pycom-libraries/blob/master/examples/lorawan-nano-gateway/otaa_node.py example it is also done in that way, so it should work to overwrite them.

choeffer, 2019-05-16:
With latest stable firmware

```
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.18.2.r7', version='v1.8.6-849-df9f237 on 2019-05-14', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')
```

it is also not working.

choeffer, 2019-05-16:
With latest developement firmware

os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.20.0.rc11', version='v1.9.4-0a38f88 on 2019-05-14', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

it is also not working.

jack17951, 2019-08-20:
(sysname='LoPy4', nodename='LoPy4', release='1.20.0.rc12', version='v1.9.4-81167ed on 2019-07-15', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')
also not working

TaliKert, 2020-02-01:
Everywhere I look, I see instructions on disabling the frequency hopping by removing all LoRa channels with something like
```
for i in range(3, 16):
    lora.remove_channel(i)
for i in range (0, 2):
    lora.add_channel(i, frequency=868100000, dr_min=3, dr_max=3)
```
Has anyone ever got this to force the TX frequency to a single frequency? So far I haven't had any luck. I tried with OTAA and ABP, firmwares ranging from 1.16, 1.18, 1.20.0 and 1.20.1 on the FiPy.

ShengLyu, 2020-08-31:
Hi, I had the [same problem](https://forum.pycom.io/topic/6303/lopy4-change-uplink-channels-by-itself) and still trying to find solutions. I created a new topic in Pycom Forum but seems nobody cares about this problem. Did you get your problem solved?